teemuatlut/TMC2130Stepper #1: Reading of stallguard-value

BlueGene00, 2017-02-28:
Hello,

I would like to read the stallguard-value like described in chapter "14 stallGuard2 Load Measurement" of the Trinamic-manual.

On youtube, I have found this, which is pretty much what I am looking for:
https://www.youtube.com/watch?v=ZAVRGcjJSv8

With your library it is possible to read DRV_STATUS. But in my tests, the value won't change if I apply load on the stepper motor. The value for DRV_STATUS only changes, when the stepper already stalled. But I want to prevent this and measure the load earlier.

Do you have already tested this feature or maybe have an idea how I can read this value properly? Are there any other settings which must be done before? I have already testet different SGT-values, but they seem not to make any difference.

teemuatlut, 2017-02-28:
I actually have tested it a bit. There were a few other values to change first to enable stallGuard. I need to get back to you a bit later this week when I get back home.

-------- Alkuperäinen viesti --------
Aihe: [teemuatlut/TMC2130Stepper] Reading of stallguard-value (#1)
Lähettäjä: BlueGene00
Vastaanottaja: teemuatlut/TMC2130Stepper
Kopio: Subscribed

Hello,

I would like to read the stallguard-value like described in chapter "14 stallGuard2 Load Measurement" of the Trinamic-manual.

On youtube, I have found this, which is pretty much what I am looking for:
https://www.youtube.com/watch?v=ZAVRGcjJSv8

With your library it is possible to read DRV_STATUS. But in my tests, the value won't change if I apply load on the stepper motor. The value for DRV_STATUS only changes, when the stepper already stalled. But I want to prevent this and measure the load earlier.

Do you have already tested this feature or maybe have an idea how I can read this value properly? Are there any other settings which must be done before? I have already testet different SGT-values, but they seem not to make any difference.

teemuatlut, 2017-03-01:
[Try with this.](https://gist.github.com/teemuatlut/2bc6d85732311087e3e5285e04c36cb5)
Use the IDE plotter to visualize the change in values.
This is my original testing scetch that I used for just about anything, and that's why there are still the tmc_write and tmc_read functions, but the writes can be replaced with methods from the library.

Let me know if it works for you.

BlueGene00, 2017-03-01:
Thanks for the example.

Yes, it is working. But only on the Arduino MEGA which I have...

My whole project ist based on a Teensy 3.6. If I want to complie the code, I get a bunch of error messages:

```
sg_value:144: error: expected constructor, destructor, or type conversion before '(' token
 ISR(TIMER1_COMPA_vect){

     ^

sg_value: In function 'void setup()':
sg_value:86: error: 'TCCR1A' was not declared in this scope
   TCCR1A = 0;// set entire TCCR1A register to 0

   ^

sg_value:87: error: 'TCCR1B' was not declared in this scope
   TCCR1B = 0;// same for TCCR1B

   ^

sg_value:88: error: 'TCNT1' was not declared in this scope
   TCNT1  = 0;//initialize counter value to 0

   ^

sg_value:90: error: 'OCR1A' was not declared in this scope
   OCR1A = 256;// = (16*10^6) / (1*1024) - 1 (must be <65536)

   ^

sg_value:92: error: 'WGM12' was not declared in this scope
   TCCR1B |= (1 << WGM12);

                   ^

sg_value:94: error: 'CS11' was not declared in this scope
   TCCR1B |= (1 << CS11);// | (1 << CS10);  

                   ^

sg_value:96: error: 'TIMSK1' was not declared in this scope
   TIMSK1 |= (1 << OCIE1A);

   ^

sg_value:96: error: 'OCIE1A' was not declared in this scope
   TIMSK1 |= (1 << OCIE1A);

                   ^

C:\Users\Dell\Desktop\sg_value\sg_value.ino: At global scope:

sg_value:144: error: expected constructor, destructor, or type conversion before '(' token
 ISR(TIMER1_COMPA_vect){

    ^

expected constructor, destructor, or type conversion before '(' token
```

Seems that the Teensy is somehow not able to work with the interrupt.

Is it somehow possible to solve this without interrupts? I tried to change the code to a simple STEP HIGH/LOW like in your other example, but then the stepper would even not get activated...

I am using the Accelstepper-library with my project, so the STEP/DIR pins are already controlled by other functions. I just want to read the stall/load value periodically and if get below a defined limit, I want to change the stepper direction, move a bit and then stop the movement. (end stop function)

teemuatlut, 2017-03-01:
The compiling is failing because the interrupts I gave there were AVR specific. For Teensy you need to use ARM specific timers like the FTM.

If you don't use interrupts, the microcontroller stepping will stop for a brief moment when it reads the values with SPI. I too had problems with readings before I changed to interrupt driven stepping.

I looked into the library you're using and it doesn't seem to be using interrupts by itself. But from what I gather from a simple example, what you could try calling the run() method from an interrupt.

Legohead259, 2018-11-06:
Hey, I've been trying to get the StallGuard example working with my set up and I can't get a reading off of it. The motor functions just fine, however, it doesn't detect the load properly. I haven't modified the example code as this is the first time I've worked with these steppers.

teemuatlut, 2018-11-07:
Either the default stallguard threshold value is not suitable for your motor or you have a hardware issue with your SPI lines.

Legohead259, 2018-11-07:
Ok. I'll try lowering it to the negative extreme and see if that works. Thanks for the response

dmitrif, 2018-11-07:
Also, the issue I found is that if you bought a clone, you have to desolder
the standalone bridge on the bottom and solder the cfg bridges.
_________________

Dmitri Farkov
647.898.5054

Legohead259, 2018-11-07:
Ok. I'll have to inspect mine then. I got the Fysetc clones so that's
probably where most of my issues are. Definitely going to limit my Chinese
purchases in the future.

Legohead259, 2018-11-07:
I've got it working now. Thank you