teemuatlut/TMC2130Stepper #2: Configuration chopper settings

KaVaLai, 2017-03-29:
Hey,

I am trying to find the right settings for a Pololu Stepper Motor NEMA 11 Bipolar 200 Steps Rev 28×45mm 4.5V 0.67 A Phase for chopper configuration but even though I calculated the parameters using your excel table and set the parameters in my arduino script using your library it did not work out... I measured the chopper cycle using an oscilloscope , measuring over the sense resistor and got a chopper frequency of round about 6kHz and a duration of the fast decay cycle much longer than tblank... I am not very familiar with the trinamic chip driver and do not know the appropriate configurations of all registers that need to be set as there are a lot. 
I tried to follow the instructions given in the pdfs that I attached. Also see my Arduino script (for a Arduino Mega 2560) and the modified excel table for my Motor.
I would appreciate a little help a lot, as far as possible!

Thanks in advance!

[AN001-spreadCycle.pdf](https://github.com/teemuatlut/TMC2130Stepper/files/878083/AN001-spreadCycle.pdf)
[AN002-stallGuard2.pdf](https://github.com/teemuatlut/TMC2130Stepper/files/878084/AN002-stallGuard2.pdf)
[TMC2130_Pulolu_Parameters.xlsx](https://github.com/teemuatlut/TMC2130Stepper/files/878085/TMC2130_Pulolu_Parameters.xlsx)
[Pololu Stepper Motor NEMA 11.pdf](https://github.com/teemuatlut/TMC2130Stepper/files/878086/Pololu.Stepper.Motor.NEMA.11.pdf)
[TMC2130_CHopper_Settings_alternative.zip](https://github.com/teemuatlut/TMC2130Stepper/files/878088/TMC2130_CHopper_Settings_alternative.zip)

teemuatlut, 2017-03-29:
Sadly I don't have easy access to a scope to do these kinds of measurements myself, so I can't see for myself.

But to get started, are you running the library from git? I recently made some fixes to the timing register methods and I've yet to push a new release to the IDE library manager.

Second, and I suspect this is why you're seeing a low frequency, I see that you're using the Arduino digitalWrite method for stepping. These functions add a lot of overhead to the commands and I would highly suggest using direct register manipulation. 

Also while this doesn't much affect your measurements, you may want to move the stepping into an interrupt routine. You'll get more accurate timings this way. I've added my personal sketch for plotting stallGuard value that uses both direct registers and interrupts.
[TMC2130_stallGuard.txt](https://github.com/teemuatlut/TMC2130Stepper/files/878193/TMC2130_stallGuard.txt)

If these don't work, then I need to start poking the registers and see if the methods are working properly.

KaVaLai, 2017-03-29:
Cool thanks!
Ran the script for my motor, but there is no change in the measured load. There are only zeros and I guess you see different values, if you start to put load on your motor... And it gets extremely hot after little time. :(

teemuatlut, 2017-03-29:
I believe the script should be in working order since I was using it just a day or two ago, but you're likely seeing zeroes because of too sensitive stallguard threshold. This setting is very motor specific. Your motor also got how because you running with my settings.

Either way, it was just to show you how I've set up the interrupts and step pin switching.

KaVaLai, 2017-03-29:
I changed the setting and played a little with the threshold... Only thing that I couldn't figure out was the cool step min speed. What does this value of 1048575 mean?
Thanks a lot for the scrpit, going to try to improve my scripts following your interrupts examle!

teemuatlut, 2017-03-29:
1048575 = 0xFFFFF = 0b11111111111111111111 = 20bit_max

Basically just the maximum possible value that fits in that 20bit register. It ensures that there is no possible value of TSTEP (and therefore movement slower) that would be greater than that. It's excessively high and another value like 100 000 would've worked just as well in that sketch.

KaVaLai, 2017-03-30:
Managed to find the right setting for the threshold, it is working out quit well!!! Only the motor is very "weak". How can I raise the torque again? I set CS to 31 but there was no difference to the torque before with a lower I_run...

teemuatlut, 2017-03-30:
Hard to say without knowing the rest. You might be overriding it with a later command. You might be running with high vsense. You might have the external voltage ref turned on.

KaVaLai, 2017-04-04:
I have a R sense of 0.11 Ohm and high v sense. External voltage ref is turned off (see file). My motors runs with rated 670mA. In the datasheet of TMC 2130 is a table for the RMS current (page 55). If I am not mistaken the current that I get with Rsense = 0.11 and vsense on lies between 1.0 and 0.9 A for RMS current. I am not sure if that is the value that I set in the TMC2130.SilentStepStick2130 command or if I have to set it to 670mA? Does this function automatically determine my CS value, if I set it to 670mA? Or do I have to enter the RMS current that I got from the table and additionally set CS? Is there a way to see the actual current value besides the SG2 value?
[TMC2130_stallGuard.zip](https://github.com/teemuatlut/TMC2130Stepper/files/892705/TMC2130_stallGuard.zip)

teemuatlut, 2017-04-04:
The current is determined by the vsense bit and the CS setting.

The library implement a setCurrent method to which you give the current is milliamps. The method will first calculate the corresponding CS setting. However, if the resulting CS setting is below 16, it will turn on the vsense bit and calculate CS setting again.

the SilentStepStick2130 does nothing more than run the setCurrent with the predefined RSENSE and 0.5 hold multiplier.

So these two methods will set both the CS and vsense bits for you.

I see you're mixing the SilentStepStick2130 method (and thus setCurrent) with the run_current and high_sense_R. So with vsense=1 (0.18V) and CS=21, you'd be looking at an rms current of  0,67311126286027120111.

teemuatlut, 2017-04-04:
You can read the CS value with run_current() or IRUN(). This will get you the CS value that is held by the shadow register and should correspond with the driver register.

KaVaLai, 2017-04-08:
Ok, so I use the setCurrent or the SilentStepStick command or the run_current and high_sense command, right? 
Other question concerning your suggestion to use interrupts to improve the accuracy of the stepper timings:
I want that my stepper only moves when he gets the command to move from a serial port entry using a If-loop searching for Serial.available(). Ergo I only what that the Interrupt that triggers the motor movement happens when the command was sent via the serial port. Is it still recommendable to use a interrupt? Because I heard that it is not recommendable to call a function to read Serial port buffer within the ISR.

teemuatlut, 2017-04-08:
Read serial in loop(). If right command is found, pull driver EN pin LOW to enable the driver.

Alternative would be to flip a boolean switch, that you then use in the ISR to check against.

Another alternative would be to enable the arduino timer when the command is received.

Lutz1967, 2018-07-24:
Hi, a different question to the same topic:
I would like to use the spreadcycle calibration example to tune noise for Marlin with RAMPS1.4.
At the end the result is supposed to be put in #define CHOPPER_TIMING xxx in Marlin configuration. But where? I coudn`t find any use of CHOPPER_TIMING alias in neither Marlin version nor in TMC library but I really appreciate the idea to put that in a configuration parameter instead of using TMC-ADV. 
How is this supposed to work?
Thanks for the great work!

teemuatlut, 2018-07-24:
The configuration option for Marlin is not upstream yet.

JackDaFlow, 2018-08-13:
Hey Lutz1967, I good starting point for the SpreadCycle settings is the ExcelSheet from Trinamic: https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC5130_TMC2130_TMC2100_Calculations.xlsx
I also modified the example file to tune more settings. You can find this file in my forked Marlin\Calibration rep. In my marlin rep I also implemented some GCode to tune the Chopper settings. I think this should be really similar to what you asked for. Hopefully this will help!

PS: There seems to be one bug when tuning fast_decay_time for the constant off time mode with the M934 command. Values higher than 4 are not accepted by the gcode. But actually I think this might be a issue in the tms2130stepper files.

teemuatlut, 2018-08-13:
> I think this might be a issue in the tmc2130stepper files

Yes. The fast decay setting has the bits split and will require some bit shifting to get it right.

JackDaFlow, 2018-08-13:
Ah, so fast_decay_time just sets f3d and not f3d&hstrt. I never looked into this. The constant off time mode did not show smoother motion than SpreadCycle, but it's louder. The only advantage is, that I could use TL-Smoothers to get some heat out of the TMC and into the Diodes.

Lutz1967, 2018-08-28:
Hi JackDaFlow, thanks for the hints. Used the spreadsheet and ended up with similar values as found with teemuatlut`s calibration example. And even with my limited capabilities I found a way to implement these values (stepper_indirection.ccp). However, there is no audible improvement, whatever values I put in. I checked with M122; correct values are reported. 
Maybe it is the hardware what is off the calibratable range ( ANET A8). So the next thing I`ll do is convert to 24V ...
I couldn`t find the GCODES you mentioned, what are they?