teemuatlut/TMC2130Stepper #3: Need help integrating into Prusa Research firmware

madhunm, 2017-05-11:
Hello,

I am trying to integrate this library into the Prusa research fork of Marlin ([https://github.com/prusa3d/Prusa-Firmware]) and am running into weird issues. 

I have basically copied over the TMC2130 declarations from Marlin 1.1.0 into the PR firmware and compilation fails with:
`C:\Users\MADHU_~1\AppData\Local\Temp\build230ef159538de2d523fd9e0a48ac3ccf.tmp\sketch\ConfigurationStore.cpp.o: In function `Config_StoreSettings()':

C:\Users\MADHU_~1\AppData\Local\Temp\build230ef159538de2d523fd9e0a48ac3ccf.tmp\sketch/ConfigurationStore.cpp:114: undefined reference to `stepperX'

C:\Users\MADHU_~1\AppData\Local\Temp\build230ef159538de2d523fd9e0a48ac3ccf.tmp\sketch/ConfigurationStore.cpp:114: undefined reference to `stepperX'

C:\Users\MADHU_~1\AppData\Local\Temp\build230ef159538de2d523fd9e0a48ac3ccf.tmp\sketch/ConfigurationStore.cpp:119: undefined reference to `stepperY'

C:\Users\MADHU_~1\AppData\Local\Temp\build230ef159538de2d523fd9e0a48ac3ccf.tmp\sketch/ConfigurationStore.cpp:119: undefined reference to `stepperY'

C:\Users\MADHU_~1\AppData\Local\Temp\build230ef159538de2d523fd9e0a48ac3ccf.tmp\sketch/ConfigurationStore.cpp:124: undefined reference to `stepperZ'

C:\Users\MADHU_~1\AppData\Local\Temp\build230ef159538de2d523fd9e0a48ac3ccf.tmp\sketch/ConfigurationStore.cpp:124: undefined reference to `stepperZ'

C:\Users\MADHU_~1\AppData\Local\Temp\build230ef159538de2d523fd9e0a48ac3ccf.tmp\sketch/ConfigurationStore.cpp:140: undefined reference to `stepperE0'

C:\Users\MADHU_~1\AppData\Local\Temp\build230ef159538de2d523fd9e0a48ac3ccf.tmp\sketch/ConfigurationStore.cpp:140: undefined reference to `stepperE0'

C:\Users\MADHU_~1\AppData\Local\Temp\build230ef159538de2d523fd9e0a48ac3ccf.tmp\sketch\stepper.cpp.o: In function `st_init()':

C:\Users\MADHU_~1\AppData\Local\Temp\build230ef159538de2d523fd9e0a48ac3ccf.tmp\sketch/stepper.cpp:662: undefined reference to `tmc2130_init()'

collect2.exe: error: ld returned 1 exit status`

Can you please help?

Thanks!

teemuatlut, 2017-05-11:
The undefined stepper references are likely due to missing include of stepper_indirection.h

Can't tell you why it can't find tmc2130_init() without seeing more of the code.

You can view my original PR to find out everything you need to get the initial driver support.
https://github.com/MarlinFirmware/Marlin/pull/5866/files

madhunm, 2017-05-11:
I have included stepper_indirection.h in ConfigurationStore.cpp, stepper.h and stepper_indirection.cpp

stepper_indirection.h has: 
```
#if defined(HAVE_TMC2130)
  #include <TMC2130Stepper.h>
  void tmc2130_init();
#endif

```

teemuatlut, 2017-05-11:
And you also have the stepperXYZ declarations in stepper_indirection.h?

If that isn't it either, then I'd have to take a closer look at the code and run it myself. Do you have it online anywhere?

madhunm, 2017-05-11:
Yes, I have the declarations for Stepper X, Y, Z and E0 in stepper_indirection.h.

I dont know how to use GitHub very well; here is a copy of the code: https://drive.google.com/file/d/0BxM5zx0gwX3mMnJKUFQ2T2VGaVk/view?usp=sharing

Let me know in case you have trouble downloading.

teemuatlut, 2017-05-11:
First remove the tmc2130_init() in Marlin_main.cpp#226. You can't use a function in that way.

Then include Marlin.h in stepper_indirection.cpp
It never compiled the code in it, because defined(HAVE_TMC2130) is false in that scope.

After that it will error when it cannot find a defined CS pin for E0. Either define the pin or comment out the 'E0_IS_TMC2130'.

madhunm, 2017-05-11:
THANKS A MILLION FOR THE ASSIST!!!

I got the file to compile. Here is the  latest copy; please review and let me know if I have made any mistakes.

https://drive.google.com/open?id=0BxM5zx0gwX3mZ3pVR3cxWTV0LW8

Thanks!

closed 2017-08-24 by teemuatlut