teemuatlut/TMC2130Stepper #6: no sg_result reading stealthChop mode

musinou, 2017-07-21:
With

`  
  TMC2130.begin();
  TMC2130.SilentStepStick2130(600);
  TMC2130.sg_stall_value(16);
  TMC2130.diag1_stall(1);
  TMC2130.diag1_active_high(1);
  TMC2130.sg_filter(0);
  TMC2130.microsteps(256);
  TMC2130.stealth_amplitude(64);
  TMC2130.stealth_gradient(15);
  TMC2130.stealth_freq(0);
  TMC2130.stealth_autoscale(0);
  TMC2130.stealth_symmetric(0);
  TMC2130.standstill_mode(0);
`

if 
`
  TMC2130.stealthChop(0);
`
then
`
  Serial.println(TMC2130.sg_result(), DEC);
`
works

but if 
`
  TMC2130.stealthChop(1);
`
then
`
  Serial.println(TMC2130.sg_result(), DEC);
`
gives 0 all the time, even if sg_stall_value is ajusted differently

teemuatlut, 2017-07-21:
StallGuard is not compatible with stealthChop. The datasheet has an explanation but basically it's a matter of current detection being different on stealthChop mode.

musinou, 2017-07-21:
I thought it could be something like this, but could not find it in the datasheet. Thank you for your reply.

closed 2017-07-21 by musinou