teemuatlut/TMC2130Stepper #10: stallGuard feature problem

kandidaten, 2017-11-07:
Hi, when i try to compile your StallGuard.ino, I get some error messages. 

'class TMC2130Stepper' has no member named 'rms_current'. 

Cant find in your description that there would be such a command in your library, only setCurrent and run_current. I changed it to run_current, which fixed it but instead i got:

exit status 1
'class TMC2130Stepper' has no member named 'sg_result'

I deleted that line and got the code running, trying out different stall values bot to no avail. I only get 0 in the serial monitor.

I have tried both your and Moritz Walters library for the TMC2130 drivers, but have not gotten stallGuard to work at all. I have tried both the StallGuard.ino file aswell as the stallguard function in the Marlin fork, tried a variety of values for THIGH and TCOOLTHRS but with no luck. I have the diag1 pin wired to the endstop signal pin on a RAMPS 1.4

Let me know if i can supply any more information to help us out to get this to work :)

teemuatlut, 2017-11-07:
THIGH and TCOOLTHRS don't actually have much to do with stallGuard.

StallGuard.ino does compile for me. Please try updating the library version.
There is rms_current and sg_result but the front page readme hasn't been updated for a bit.

kandidaten, 2017-11-08:
Okey, I must´ve misunderstood the datasheet. This line made me think the TCOOLTHRS is relevant for getting the stallGuard function to work:

"TCOOLTHRS ≥ TSTEP ≥ THIGH:
- coolStep is enabled, if configured
- stealthChop voltage PWM mode is disabled
TCOOLTHRS ≥ TSTEP
- Stop on stall and stall output signal is enabled, if
configured"

I will update the library and try again, thanks :)
Great job creating this library!

teemuatlut, 2017-11-29:
Do you have the original Watterott made drivers or the ones that are sold on Aliexpress, etc?
It's recently come to my attention that the "clones" don't necessarily work the same and just last week I tried the stallGuard example which worked perfectly for me. And I am using Arduino Mega and RAMPS1.4.

A value of 1023 sounds like a communication problem because in BIN it's all ones.

kandidaten, 2017-11-29:
Yes, the original Watterot drivers on X and Y axis. I can put them into stealthChop mode via the Marlin code, which indicates the SPI works as it should, right? 

I have also tried to measure on the diag1 pin to detect a pulse upon stalling, but have not detected anything. Maybe the pulse is too short.

kandidaten, 2017-12-05:
Have I understood it corectly if I presume SPI-interface works fine if stealthChop can be turnet on/off via the Marlin code?

kandidaten, 2017-12-05:
Tried your StallGuard sketch again. I upload the sketch, and the motors starts to turn. It says "0 1023" no matter the load on the motor. When I turn the 12V source off, it immidiately changes to "0 0", turning it on again lead to "0 1023". 

When you say communicatons error, what should i look for?

Thanks in advance :)

teemuatlut, 2017-12-05:
Yes, if you can turn on or or off the stealthChop bit, your communication should be ok. At least the TX line.

kandidaten, 2017-12-06:
Okey, thanks. Sorry for bombarding you with questions here. You say the fact that I only see "0 1023" when i run the StallGuard sketch indicates some kind of communications error. Where would that be?

teemuatlut, 2017-12-06:
I don't expect you'd be able to find answers from elsewhere.
I reckon you've probably done this already multiple times but check your wiring. The 1023 may be a caused by a loose connection. If only the TX line is faulty, you'd still be able to configure the driver but will have incorrect responses.
You can further test this by adding this into your setup()
```cpp
Serial.print("DRV_STATUS=0x");
Serial.println(TMC2130.DRV_STATUS(), HEX);
```