teemuatlut/TMC2130Stepper #11: Inconsistencies in stepper motor response when using stallGuard

beccathermo, 2017-11-09:
I am working on a project in which a stepper motor moves a heavy load along the z-axis. There are bearings on each end that I use stallGuard to "detect" and have the code change the direction of the stepper motor. While I have used this library successfully for a simpler application (using stealthChop on a stepper with minimum load), I continue to experience inconsistencies in the motor's response when using stallGuard.

Setup:
- 12V/5A power supply
- 2.34V phase voltage
- Teensy 3.2
- TMC2130
- 600mA rms current
- heat sinks/fan used

For example, without coding any of the settings for dcStep, it will inconsistently "turn on" when moving in the negative z-direction. It will also slip-causing the load to slide down the shaft for one second-when changing direction from positive z to the negative z, but it only does this occasionally. (There is a clear difference in the slip and dcStep.) One other example is that each time I compile code or hit the reset button on the Teensy, the motor stalls briefly after one second before doing what I told it do. Is it possible it's calculating something during that time?

I can compile and upload the same code two times and it will behave differently. Do I need to reset the parameters each time I upload the code? Any thoughts?

teemuatlut, 2017-11-09:
I'd recommend a full power cycle between compiles if you see it acting weird. The registers may not get a proper reset on start up and I remember having some trouble with resets at some point.

It's hard for me to point anything clear but
* Remember that stealthChop is not compatible with stallGuard. Which I presume you already know.
* The stallGuard value will depend on the motor current
* Randomly triggering stallGuard may be because of insufficient current. Basically that 600mA may just be enough to drive the load but on the edge of skipping a step.
* The library doesn't add anything calculation intensive but there are things that are part of the core so it may take a moment to actually get to your sketch.

beccathermo, 2017-11-09:
The full power cycle didn't improve the consistency unfortunately.

I also upped the current to 1A and still face the same issues. I've reached out to Watterott (supplier of the SilentStepStick motor driver I am using) and will provide an update if I reach any major breakthroughs.

Thank you for the work you've done with this library!