teemuatlut/TMC2130Stepper #12: StallGuard example on nano

kevoc69, 2017-12-06:
Hi,

thanks for you library and examples.
I'm trying to get your StallGuard example to work on a arduino nano
but (obviously) PORTF isn't defined. Actually, I'm only just learning about 
port manipulation so I'm still a little unsure what your are trying to achieve 
with 

ISR(TIMER1_COMPA_vect){
  PORTF |= 1 << 0;
  PORTF &= ~(1 << 0);
}

is this setting the STEP pin?
Any help would be gratefully received.
Best regards,
Kevin

teemuatlut, 2017-12-06:
PORTF is the Arduino pins 54..61.
Basically I'm writing (bitwise OR) a one to the first bit position (0) of the pin register to toggle it high. And then I drive it low again with bitwise inverse AND.

For nano you could use pin 18 which would be PORTC and bit position 4. Search for Arduino Nano pinout and look for PC4.

kevoc69, 2017-12-06:
thanks for the clarification - much appreciated.

closed 2017-12-06 by kevoc69