teemuatlut/TMC2130Stepper #13: Direction issue

fhendrikx, 2017-12-12:
Hi,

Firstly, thanks for your work on this library, I'm excited to be using this new driver. 

I'm hoping you can help with this odd situation. I've just switched from using a DRV8825 driver to the TMC2130. The simple example mostly works for me, except I can't seem to get it to change direction (I did need to change the microsecond delay from 10 to 100, otherwise the motor just stalls). I've double checked the DIR pin configuration and wiring, so I'm kind of at a loss. Any thoughts on next steps or things to consider?

Thank you.

teemuatlut, 2017-12-12:
Do you have the original Watterott drivers or one of the clones?
My first thought is that the drivers are in stand alone mode and that's why they work but communication isn't working. Did you solder the SPI jumper bridge? It should be left open.
You shouldn't need to change the delay for the drivers to work.

fhendrikx, 2017-12-12:
Thanks for your response. I got the driver from aliexpress; imagine it's probably a clone (?) The driver is white, "TMC2130 V1.0 WWW.FYSETC.COM" printed on it. I haven't modified the driver at all.

teemuatlut, 2017-12-12:
Please see [this](https://github.com/MarlinFirmware/Marlin/issues/8480#issuecomment-350824499).

fhendrikx, 2017-12-12:
Looks like the SPI bridge has a join, and the others are not joined. If I understand correctly, I should remove the SPI bridge, and add the missing CFG4 and 5 links?

[image]

teemuatlut, 2017-12-12:
Comparing with the image by @nightdemonx it would seem so. I don't personally have any of the Fysetc drivers.

nightdemonx, 2017-12-12:
> I should remove the SPI bridge, and add the missing CFG4 and 5 links?
Yes, you right! Chines produce his version for stand alone using.

fhendrikx, 2017-12-13:
Thanks for your feedback. I tried making the changes to a driver today, no luck. The motor doesn't get energised anymore at all. Will try another this weekend.

nightdemonx, 2017-12-13:
I not undersrtand, how you want connect driver. If it yours photo of driver, you dont want use SPI and you must solder SPI bridge. If you want use SPI, you must also connect CSN, SCK, SDI, SDO to Mega, leave SPI bridge open and soldering CFG4,5.

fhendrikx, 2017-12-13:
I want to use the driver with teemuatlut's TMC2130Stepper code. This code uses the Enable, CS, STEP and DIR pins. However, the driver wasn't responding to the DIR pin (it was always going counter-clockwise), so I asked teemuatlut and he said that the SPI bridge must be open as per your thread. However, now that the SPI bridge is open, the driver doesn't work at all.

nightdemonx, 2017-12-13:
Teemuatlut's TMC2130Stepper code work only if you use SPI mode (connect CSN, SCK, SDI, SDO to Mega, leave SPI bridge open and soldering CFG4,5). Do you connect 4 pins to mega? 
[image]

fhendrikx, 2017-12-13:
Of the 4 SPI pins you mention, I only have CSN connected. The 4 pins I have connected to the arduino are: Enable, CS, STEP and DIR. These four are mentioned in the constructor:

  #include <TMC2130Stepper.h>
  TMC2130Stepper TMC2130 = TMC2130Stepper(EN_PIN, DIR_PIN, STEP_PIN, CS_PIN);

If more pins are required, then I'm not sure how the class would know about them. Maybe I'm missing something here?

Thanks!

teemuatlut, 2017-12-13:
You also need the hardware SPI pins; MOSI, MISO and SCK. Because these are hardware, they are defined by the Arduino core.

teemuatlut, 2017-12-13:
Actually I will be adding software SPI at some point and the EN, DIR and STEP will be removed from the constructor. The library doesn't make use of these pins anyway.

fhendrikx, 2017-12-13:
Ah, that makes more sense. So, can you tell me what pins I actually need to make it work? In particular, which pins on the arduino to go which pins on the driver module?

On my UNO, SPI pins are 13 (SCK), 12 (MISO), 11 (MOSI) and 10 (SS).

teemuatlut, 2017-12-13:
You can refer to [this](https://github.com/watterott/SilentStepStick/blob/master/hardware/SilentStepStick-TMC2130_v10.pdf) for the driver pinout. Connect the wires but the CS (or SS) can be any digital pin.

wbarber69, 2018-03-18:
For anyone wanting to purchase the Fystech drivers for use in spi mode, when searching for them on aliexpress look for version 1.1

closed 2018-05-01 by teemuatlut