teemuatlut/TMC2130Stepper #14: "Short to ground" errors in Marlin 1.1.7

mbuc, 2017-12-29:
Hey @teemuatlut, I really appreciate the work you have put into providing this library.

I'm having an issue with my authentic TMC2130's.  They didn't perform all that well in stealthChop mode, and after turning hybrid mode off in Marlin and switching to SpreadCycle, it printed smooth as ever.  Then I started getting the following error as soon as I turn on my power supply:

> Y driver error detected:
> overtemperature
> short to ground (coil A)
> short to ground (coil B)

I've checked the motor - nothing is shorted to ground. Each winding pair connects to each other, and none are shorted across windings. The resistance is low and equal among both windings (and match the X motor exactly). I've checked and redone the wiring, nothing is amiss.

It's not overheating as the heatsink is cool and the fan is on.

I've swapped the stepper drivers and gotten the same error message for Y. I then left them in the same places, but swapped the chip select between them - and still, "Y driver error detected...".  If I unplug both chip select cables, I get:

> X driver error detected:
> overtemperature
> short to ground (coil A)
> short to ground (coil B)

This makes me wonder if SPI is getting interrupted partway through or something.  Any ideas?  Am I completely off base in thinking this is a software issue?

mbuc, 2017-12-29:
Upon further inspection, it looks like the chip select pin for the Y driver is floating - perhaps there's been some damage to the Arduino?  It's an official 2560 board from Italy.

Attached is a screenshot of the overview of my logic analyzer.  Top half is digital, bottom half is analog.  Looks like a floating pin to me. 

[image]

teemuatlut, 2017-12-29:
You could try another pin to confirm your theory.

The driver seems to report all errors at the same time, which is very unlikely when working correctly. This makes me think there's a hardware communication issue and the response from the driver was 0xFFFFFFFF. This is likely to happen when the wires aren't connected properly or the CS pin doesn't work right.

I will be adding a "faulty communications" warning to Marlin when I get around to it.

mbuc, 2017-12-29:
I tried swapping pins with the chip select for the E extruder in pins.h, but that led to no output on either the new pin (D44) nor the default Y chip select pin (D49).  I only have TMC2130's on the X and Y driver sockets, and A4988's in Z, E0, and E1, so that's confusing behavior to me.  I'll see if I can't get some more advanced debugging going before making any more assumptions.

mbuc, 2018-01-04:
I've replaced the RAMPS board and tried using Pin 4 instead of Pin 49 for chip select.  Both TMC2130 boards are working fine but no matter what the chip select isn't becoming active for Y.

mbuc, 2018-01-04:
Apologies, edited the wrong file.  Edited in a different version's directory.  Changing Y_CS_PIN to 4 finally worked.

closed 2018-01-04 by mbuc