teemuatlut/TMC2130Stepper #20: TMC2130 Not Working/Powering Question 24V w/M80,M81

CHILLYSMOKES, 2018-02-17:
Ramps1.4/Mega2560
24V mod
POWER_SUPPLY 1 (M80,M81)
Marlin1.1.8
Octoprint
TMC2130 X,Y 
DRV8825 Z,E

Sorry for lengthy story but I'm pulling my hair out and I've googlized a bunch. Thanks for your work.
I can't get any stepper, with a TMC2130 driving it, to move with "my setup". Installed with Tom's Instructions and rechecked wiring many times. Removed TMC2130s and re-connected DRV8825s and steppers work. Put TMC2130s back in and nothing. Uploaded the example provided, here, and realized I have no way to power Ramps using M80,M81 through the example sketch. Dug out 12V PSU and connected power to Ramps and no movement (previously uploaded the example sketch). With 12V PSU still on, uploaded example sketch and VOILA!! X axis movement. Very harsh vibrating, definitely not good. Commented out "stealthchop" in example then re-uploaded and movement was definitely worse. Played with motor current in example sketch and found 800mA to be best but still horrible vibrations. Easilly able to reverse x-direction by pushing carriage in opposite direction. Looks like all communication is functioning with the example sketch but not with Marlin. I've checked pins_Ramps.h and its correct. Changed _CS to 4 and 5 and rewired. Nothing. M122 S1 shows register/hex changes when I try to home but no movement.

24V vs 12V issues??
Do the TMC2130 drivers need power at the time of Mega2560 boot to function?
Any help at this time is appreciated. Thank you.

CHILLYSMOKES, 2018-02-18:
Continuing troubleshooting results in no more vibration.  Smooth as silk. The powering is still an issue.

bustedparts, 2018-02-22:
Sorry buddy. I just seen this ! I will be working on these drivers again
really soon again. I just got a new set.  Also briefly reading what you
have wrote. You have to make sure when plugging in a USB to upload a new
sketch to your mega. You HAVE TO have the power to the ramps board on! If
you fail to do this you will damage the chip on the stepper drivers.

teemuatlut, 2018-02-22:
Marlin v1.1.8 has a bug that affects boot time configuration of the TMC2130 drivers. Use the bugfix-1.1.x branch instead.

CHILLYSMOKES, 2018-02-22:
Thanks guys!!
That explains a lot. I really appreciate your time. Was already working on a fix that will, coincidentally, address both your comments. Instead of using Ramps PS_ON to trigger my PSU relay, I'm gonna use the RPi GPIOs with a darlington chip to control both power supplies, 24V and 9V. Then I wont be mucking about with the USB cable. Thanks again.

CHILLYSMOKES, 2018-02-22:
Hopefully this will help someone else.

closed 2018-02-22 by CHILLYSMOKES