teemuatlut/TMC2130Stepper #23: Pin numbering unclear (?) - Arduino nano

LordZnarf, 2018-03-08:
Hey,

don't know if it's just me being to stupid, but I think I can't find the right pins to connect everything. I would love a Fritzing scheme, but as long as it's not done: What pin numbering do you use?

There are these: https://www.likecircuit.com/wp-content/uploads/2017/04/arduino-nano-pinout.jpg
and others: http://www.pighixxx.com/test/wp-content/uploads/2014/11/nano.png

Thanks for your help :)

teemuatlut, 2018-03-08:
It's been a very long time since I've tried the drivers on a nano, but try this:

TMC2130 | Nano
---:|:---
SDI | MOSI or D11
SDO | MISO or D12
SCK | SCK or D13
CS | D17 / A3
EN | D16 / A2
DIR | D19 / A5
STEP | D18 / A4

LordZnarf, 2018-03-08:
Okay, thanks, I'll try that! :)

LordZnarf, 2018-03-08:
Well the pins worked... Thanks!

Unrelated to the pins I just blew my driver when I turned up the voltage 😢

closed 2018-03-08 by LordZnarf