teemuatlut/TMC2130Stepper #27: marlin 1.1.8 and TMC2130 stepper library 2.2.0 get compilation error

Hywelmartin, 2018-03-17:
Error code from Arduino IDE 1.8.5

Build options changed, rebuilding all
In file included from sketch/Marlin.h:46:0,
                 from sketch/Marlin_main.cpp:247:
sketch/Marlin_main.cpp: In instantiation of 'void tmc_status(TMC&, TMC_AxisEnum, TMC_debug_enum, float) [with TMC = TMC2130Stepper]':
sketch/Marlin_main.cpp:10391:73:   required from here
serial.h:37: error: 'class TMC2130Stepper' has no member named 'hysterisis_end'
   #define MYSERIAL customizedSerial
                    ^
sketch/Marlin_main.cpp:10366:24: note: in expansion of macro 'MYSERIAL'
         case TMC_HEND: MYSERIAL.print(st.hysterisis_end(), DEC); break;
                        ^
serial.h:37: error: 'class TMC2130Stepper' has no member named 'hysterisis_start'
   #define MYSERIAL customizedSerial
                    ^
sketch/Marlin_main.cpp:10367:25: note: in expansion of macro 'MYSERIAL'
         case TMC_HSTRT: MYSERIAL.print(st.hysterisis_start(), DEC); break;
                         ^
exit status 1
'class TMC2130Stepper' has no member named 'hysterisis_end'

[setup.zip](https://github.com/teemuatlut/TMC2130Stepper/files/1821475/setup.zip)

Saur0o0n, 2018-03-17:
I can only confirm the problem with my setup...

teemuatlut, 2018-03-17:
This is because a recent bugfix required changes to both Marlin and the libraries.
You need to either downgrade the library or upgrade Marlin to bugfix-1.1.x branch.

Saur0o0n, 2018-03-17:
Well, yep it seems to fix the problem - I had two days old bugfix branch ;)

Hywelmartin, 2018-03-17:
Ok so it's sorted out.... closing

closed 2018-03-17 by Hywelmartin