teemuatlut/TMC2130Stepper #28: MKS-Gen L V1.0 w/5 Fystech 2130 drivers for reference

wbarber69, 2018-03-18:
[image]
here is a rudimentary drawing of how im using my drivers on an MKS-Gen L V1.0 board.  Just in case anyone else needs a refernce

Budavaril, 2018-03-20:
Hi,
Why better to use those irregular pins (example servos pins) than using same standard pins (aux2-aux3)on any other ramps based boards?

wbarber69, 2018-03-20:
In bugfix branch of marlin software serial has been implemented.  And is also recommended by the marlin guys.  By using software serial you remove the traffic over the hardware serial line which can cause connection issues over usb on ramps based boards.  Also this allows the full use of both the full graphic lcd and the sd card pins without having to hack or splice into anything.  Pins 4 and 5 of the servo have been recommended by other ramps users so I just followed that line of thinking and kept my connectors simple, since I am using 5 steppers in spi configuration.   I used pin 65 because it’s close to servo pin and also because I had already made my spi connector for the aux3 pinout so I just moved it over to the aux2 connector keeping the same pinout.

Budavaril, 2018-03-20:
That make sense. So may that's why I had a some trouble with my FLsun i3 during TMC2130 upgrade?
 I using my board with "dual screen" . 
One of the 3.2" touch screen has SD and WiFi built in, that's why I thought if I using for "second screen"  just an (Anet Discount) 12864 what has no SD feature I should have no any those problems to changing my drivers to TMC2130. 
But may yes .. Because I tried first using Marlin v1.1.8 and library TMC2130 v2.2.1 but that combo wasn't succesful. Motors just moved shortly with strong noise and I got few messages:
"00:00: X driver overtemperature warning! (0mA)"
"X driver error detected: short to ground (coil A)"
"Error:Printer halted. kill() called!"
I thing there is some bug and the "AUTOMATIC_CURRENT_CONTROL" (What called "MONITOR_DRIVER_STATUS" in v1.1.8) does nor really worked and my motors got (uncontrolled) maximum current.
After I downgraded to Marlin 1.1.6 and TMC2130 2.0.0 looks everything working. But I still have to troubleshooting because I still loosing some steps on X and Y. 
I thought that can be solving with some "current-speed-acceleration tune". But may there is a communication problem?

wbarber69, 2018-03-20:
I was going to do the same thing but octoprint can no longer connect reliably with the mks screen attached and the lcd. So I scrapped the touch screen for a RepRap discount full graphic lcd and haven’t had an issue since.

wbarber69, 2018-03-20:
If you are still using hardware serial then you are flooding the serial line with information.  I’ve found that most of the x any y skipping I’m having has been caused by inadequate cooling of the drivers coupled with fast acceleration/velocity.  I can cause the skips right now by either enabling interpolation or steps or by either increasing my acceleration or velocity.  I am currently installing some stack headers and more heatsinks and am trying to mitigate excessive wires blocking airflow.  I should be able see some improvement by this weekend so I will update once I get my thermal tape in the mail.

wbarber69, 2018-03-20:
Also I had this issue with the overtemp warning at first as well.  What ended up fixing that was to make my wiring harness so that each serial line went directly to the spi pins instead of daisy chaining them.  Apparently the spi bus cannot handle more than one device if they are wired up like this. So I threw out my first harness and instead opted for twisting all the wires together and capping them with some heat shrink.  I intend to design a better plug, maybe a small pcb to sit on top of the pins I need and then breakout the wires from there.  So to reiterate, if you have your spi wires going first to one stepper then to the next stepper you will need to change it so that all of the wires from each individual stepstick go directly to the spi header.

teemuatlut, 2018-03-20:
Marlin v1.1.8 has an issue with setting the pin mode for CS too late and that causes problems with the configuration. The bugfix-1.1.x has this fixed.

Also it might be clearer to use "serial" to mean "USART" and not "Serial Peripheral Interface".

wbarber69, 2018-03-20:
is that correct though? every where its referenced that tmc2130 is connecting in spi mode. 2208's use a uart connection.

Budavaril, 2018-03-22:
I tried using the bugfix-1.1.x  (not enabled the SW serial yet)and  steppers are working looks well, but if I start a print (from pc or from the touchscreen's SD-USB) the bed and hotend doesn't heating up, just  start immediately moving the head looks printing.
Any idea what I missed?

wbarber69, 2018-03-22:
Double check your configuration file.  You might have I commented or accidentally deleted something.  Try comparing your files to an example for a similar machine and make sure nothing is out of whack.  Also make sure you power cycle everything just to make sure it’s not a random glitch.

Budavaril, 2018-03-22:
Thanks your quick response.
So it was strange. I uploaded previous 1.1.6 version what was working before, but got same problem. 
I removed the USB cable, restarted the printer after start printing from SD card normally (first 1.1.6 and after bugfix1.1.x too). 
I connected my laptop again, but same problem came back. 
Restarted printer and laptop, after working everithing. 
So was something problem with the laptop USB but firmware upload was working. 
I don't understand, but most important this part have fixed. 
I still loos steps, I don't know what I should changing first.. The cables to same lenght (no chain) or change pins and activate SW serial? Finally I want to do both, but I'm courious which one cose in my configuration at the moment.

wbarber69, 2018-03-22:
images of the changes ive made to my step sticks to try to ensure better cooling.
Ive added heatsinks directly to the chip on the bottom an bent the spi header to facilitate better airflow when the dupont connector is inserted.  I am using some arduino stack headers i got in a multipack to raise the chips up off the board.
[image]
[image]
[image]

wbarber69, 2018-03-22:
And with everything put back together.  I may clip those stack headers down in the future.  Currently Ive re-enabled stepper interpolation and am running my current at 650 650 650 700 700.  Im printing a nice medium sized part with a lot of circular patterns so I can see if i get any more layer shifting.  so far it is nice and quiet and running pretty smooth.  only time will tell.

[image]

wbarber69, 2018-03-23:
Definitely take your steppers out of series.  Look at my recently posted pictures.  If you don’t have plenty of cooling on your step sticks you will start to lose steps.  Definitely try out software serial. And if all else fails go into your firmware under the tmc settings and change interpolation to false.  This drops it back down to 16th stepping and is what I have been running until I made the recent changes.

wbarber69, 2018-03-23:
I had my first successful print with stealthchop and interpolated stepping.  Still some fine tuning to do on my slicer side of things but I printed for 5 hours straight at 60mm/s with 30% fast honeycomb infill on a part that is all curved outer perimeters with a separate thin but tall tower and not a single layer shift can be seen.  So I can definitely say that cooling is the first step to a successful install. I also noticed that my motors are not getting near as hot as they were even when I was working with plain old 16th stepping.

Budavaril, 2018-03-23:
[image]
I still try all adviced upgrade, but not better.
I tried cooling with a 5015 blower fan from chip side, other 2pcs 40mm from "cable side) but result is same.
Another intresting, the lose steps coming on same points,most of the times between two layers.
Difference between the results, this effect sometimes just on X or on Y, sometimes on both axis coming.
Happening always when accelerating in travel (non printing) and motor(s) give "scratchy" noise. 
I can do same if "playing" with the motors, I mean controlling manually by laptop (usb) or from any own (display) controller, if moving 10mm steps fast after each other. No matter, same direction or change direction. About every 10-15 movment give that noise and after of course able to go over the end points because lost the correct home position. 
Can I adjust "jerk" or "acceleration" in config.h or if TMC2130 used there is another way?

wbarber69, 2018-03-23:
From your pictures I would say your drivers are definitely shutting down from over overheating try bringing your acceleration down to 100 and velocity under 500.  I would recommend the heatsink mod as I got immediate positive results.  Another thing you can try is to increase your psu power from 12v to 14v.

[20180323_172638]<https://user-images.githubusercontent.com/19350571/37844545-158c08b4-2ec0-11e8-944e-c023c322e990.jpg>

Budavaril, 2018-03-23:
Ok, I have 5pcs brand new "old type" driver and those heatshinks doesn't fitted yet, so I can using them instantly on the chips. Let's try.. 
I just don't understand, if overheating happening, why I can't see that in the log? 
I guess any of this error message should be show there same than I got this message with official 1.1.8 which is doesn't adjust the current and after 10mm movement sent message "00:00: X driver overtemperature warning! (0mA)" 
Anyway my 12V -> 24V step up converter and the timer modul are alredy on the way for that upgrade. May that will be solving? Until of course I can try to adjust my 12V PSU..
I still courious, how working instantly on the "old" Ramps boards (example in Thomas Sanladerer's video) without extra cooling and why we have this problem on MKS GEN L? What's the difference?

wbarber69, 2018-03-23:
I was having those same overtemp errors with my wires daisy chained.  Also you can turn off the monitoring to keep it from shutting down during those messages. But then your only way to know there is a problem is if you have later skipping.  The reason why you think Thomas sandlanderer didn’t have any problems is because he never went any further into the issues.  Go to his website and you’ll see his forum has become inundated with people just like you and me that are having these issues.  That’s how I found to use the servo pins instead of the lcd pins

wbarber69, 2018-03-23:
All of the modifications I have posted have been recommended by the writer of this library.

Budavaril, 2018-03-23:
Now cooled 2side by heatshinks. Uppers blowed 2pcs 40mm fans. Unders (on chip) blowed with 5015.
Voltage 13.9V (that's the maximum) 
in firmware:
#define DEFAULT_MAX_ACCELERATION      { 100, 100, 50, 10000 }
#define DEFAULT_ACCELERATION          100    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  100    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   100  // X, Y, Z acceleration for travel (non printing) moves
velocity on screen:
Vmax X:100
Vmax Y: 100
Vmax Z: 2
Vmax E: 25

Result: 
[image]

At the moment finished my all ideas..

teemuatlut, 2018-03-23:
To clear up a few things
TMC2208 uses Serial
TMC2130 uses SPI

There is no communication from the motherboard to the driver after the initial boot time, unless you have `MONITOR_DRIVER_STATUS` enabled.
It doesn't matter if you use SW SPI or HW SPI. The only difference is pins used and HW SPI is faster.
It shouldn't matter if the MOSI/MISO/SCK wires are from one driver to another or all starting from the same point. Electrically this is the same.

"Scratchy" noises from the motor indicate a mechanical problem.

An acceleration of 100 would be excessively low. Max velocity of 500 would be excessively high.

Please post your M503 and M122 output if you're seeing skipped steps.

wbarber69, 2018-03-23:
It shouldn’t matter how the wires are tied in, but it does.  It’s very hard to get 2 wires into one of those crimpable style connectors.  And it was the first problem I encountered and it was almost the first google entry that led me to realize that one bad crimp can damage the whole harness and when I did wire up in series I got the driver overtemp on x no matter what driver I plugged in. The first solution I found was to try parallel using some factory duponts.  All I did to remedy this issue was to twist all of them together and heatshrink after that it was on! I hate the cheap crimp style connectors. And it’s the first place to troubleshoot imho.  I used some single ended DuPont ribbon cables I had lying around and removed them from the single connectors and applied them to whatever connector I needed.  It’s quick and dirty and I plan to route out a simple pcb to make everything cleaner.  But my layer skipping is now a thing of the past.  Definitely turn off the monitoring if you have a stable setup.

Budavaril, 2018-03-24:
Thanks all of that, now I understanding better how works.
First my outputs:
 M503:
11:14:37.492 : echo:  G21    ; Units in mm
11:14:37.493 : echo:  M149 C ; Units in Celsius
11:14:37.495 : echo:Filament settings: Disabled
11:14:37.496 : echo:  M200 D1.75
11:14:37.496 : echo:  M200 D0
11:14:37.497 : echo:Steps per unit:
11:14:37.499 : echo:  M92 X100.00 Y100.00 Z400.00 E150.00
11:14:37.501 : echo:Maximum feedrates (units/s):
11:14:37.503 : echo:  M203 X100.00 Y100.00 Z2.00 E25.00
11:14:37.504 : echo:Maximum Acceleration (units/s2):
11:14:37.506 : echo:  M201 X9000 Y9000 Z50 E10000
11:14:37.510 : echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
11:14:37.511 : echo:  M204 P3000.00 R3000.00 T3000.00
11:14:37.516 : echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
11:14:37.518 : echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.30 E5.00
11:14:37.519 : echo:Home offset:
11:14:37.520 : echo:  M206 X0.00 Y0.00 Z0.00
11:14:37.522 : echo:Auto Bed Leveling:
11:14:37.522 : echo:  M420 S0
11:14:37.523 : echo:Material heatup parameters:
11:14:37.524 : echo:  M145 S0 H180 B50 F0
11:14:37.526 : echo:  M145 S1 H240 B60 F0
11:14:37.526 : echo:PID settings:
11:14:37.527 : echo:  M301 P22.20 I1.08 D114.00
11:14:37.528 : echo:Z-Probe Offset (mm):
11:14:37.530 : echo:  M851 Z-0.50
11:14:37.531 : echo:Stepper driver current:
11:14:37.533 : echo:  M906 X 800 Y 800 Z 800 E0 800 E1 800

and M122:
11:17:04.013 : X	Y	Z	E0	E1
11:17:04.015 : Enabled		false	false	false	false	false
11:17:04.016 : Set current	800	800	800	800	800
11:17:04.021 : RMS current	795	795	795	795	795
11:17:04.027 : MAX current	1121	1121	1121	1121	1121
11:17:04.029 : Run current	25/31	25/31	25/31	25/31	25/31
11:17:04.031 : Hold current	12/31	12/31	12/31	12/31	12/31
11:17:04.034 : CS actual		12/31	12/31	12/31	12/31	12/31
11:17:04.040 : PWM scale	2	2	2	2	2
11:17:04.043 : vsense		1=.18	1=.18	1=.18	1=.18	1=.18
11:17:04.047 : stealthChop	true	true	true	true	true
11:17:04.051 : msteps		16	16	16	16	16
11:17:04.055 : tstep		1048575	1048575	1048575	1048575	1048575
11:17:04.056 : pwm
11:17:04.056 : threshold		0	0	0	0	0
11:17:04.060 : [mm/s]		-	-	-	-	-
11:17:04.062 : OT prewarn	false	false	false	false	false
11:17:04.062 : OT prewarn has
11:17:04.067 : been triggered	false	false	false	false	false
11:17:04.071 : off time		5	5	5	5	5
11:17:04.071 : blank time	24	24	24	24	24
11:17:04.076 : hysteresis
11:17:04.076 : -end		2	2	2	2	2
11:17:04.079 : -start		3	3	3	3	3
11:17:04.080 : Stallguard thrs	0	0	0	0	0
11:17:04.080 : DRVSTATUS	X	Y	Z	E0	E1
11:17:04.087 : stallguard
11:17:04.087 : sg_result		0	0	0	0	0
11:17:04.093 : fsactive
11:17:04.093 : stst		X	X	X	X	X
11:17:04.102 : olb
11:17:04.102 : ola
11:17:04.102 : s2gb
11:17:04.111 : s2ga
11:17:04.111 : otpw
11:17:04.111 : ot
11:17:04.113 : Driver registers:	X = 0x80:0C:00:00
11:17:04.115 : Y = 0x80:0C:00:00
11:17:04.117 : Z = 0x80:0C:00:00
11:17:04.117 : E0 = 0x80:0C:00:00
11:17:04.119 : E1 = 0x80:0C:00:00

teemuatlut, 2018-03-24:
See M201.

Budavaril, 2018-03-24:
Wow.. No output.. Possible??

teemuatlut, 2018-03-24:
In your M503 output you just posted.
You're using acceleration values of 9000.

Budavaril, 2018-03-24:
That was the original and last time just set everything back to original.
But was same (bad) result if I adjusted  like I posted yesterday:
#define DEFAULT_MAX_ACCELERATION { 100, 100, 50, 10000 }
#define DEFAULT_ACCELERATION 100 // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION 100 // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION 100 // X, Y, Z acceleration for travel (non printing) moves
velocity on screen:
Vmax X:100
Vmax Y: 100
Vmax Z: 2
Vmax E: 25

teemuatlut, 2018-03-24:
M503 shows your current settings.
The set default values in configuration.h are overriden by the values stored in EEPROM.

Budavaril, 2018-03-24:
Oh yeah... I think that was the key.. 
Now I set all this value to 100 on screen, and  printer became like old man. So that's why I couldnt realise any change about the movements any time... I tried to set all that values to 10 and had no any visible effect..  
So good news because I have my original board "untached" (Just in case I ordered another same board to do all my experiments not on my original working board.) I can connect and read the original values after this "slow-motion" printing..

teemuatlut, 2018-03-24:
Just send `M201 X1000 Y1000` or use the LCD to update the values on an on going print.

Budavaril, 2018-03-24:
Ok. that's cool. Now became "younger" and to be honest movements more similar how was with the old drivers.. Printing nearly finish and no any skipped steps until now. :)

wbarber69, 2018-03-24:
Sorry, I realize I had acceleration and velocity settings backwards earlier when I made the recommendation. Lots of settings and numbers to keep up with.  This mornings tests are going well.  I have doubled all my values for speed and acceleration and so far (fingers crossed) no layer skipping.  Hopefully this trend will remain and I can get back to decent printing speeds.  My first test at 60mm/s with slow settings was successful but outer perimeters were kinda not where I wanted them, but at 40mm/s we’re almost perfect.

wbarber69, 2018-03-24:
so it seems that acceleration is kind of flexible, as it doesnt seem to make as much of an impact as velocity does.  im stable at or below 150-175.  but i have tried accelerations of up to 1000 I havent gotten around to testing any higher values yet.  my first test was with x and y at 250 with acceleration of 1000.  I was also using a high jerk setting of 30.  i had failures at the first sign of tighter shorter movements. so when my base feature was finished and only wall type features were being printed almost every other layer was problematic.  stopped and went back to original velocities but increased acceleration had no problems.  so ive begun to tune velocity only.

Budavaril, 2018-04-09:
I received the parts for 24V (12V/24Vstep-up converter and timer modul), but I'm not sure about the delayed method. First powered the MKS GEN L 12V and after the drivers "VM" 24V or first the 24V after the board 12V?

teemuatlut, 2018-04-09:
If you can decide, apply the 24V first.

qavro, 2019-04-05:
Hello. Somehow i think you might be my savior, so i apologize in advance.
I have an Ender 3. After i have tunned the mechanical part next to "God" mode ( i am really advanced in CNC precision machines) i am ready to move on the brain. I have bought MKS GEN L board with regular A4988 drivers. I intend to change the drivers but that is another story, the board simply came with these. I have configured and  installed the Marlin 1.1.9 (fixed) with no issues. But here comes my problem. I really want to keep my Ender 3 display. No biggie, simply works with the board after i swap the ribbon cable around. But i really really need and SD card. So, i bought the MKS SLOT V1.1. Can you please help me to understand how can i connect it to the board and/if is there anything else i should edit in my configuration file?

BIG thanks to you or anybody willing to give me a hand here.

teemuatlut, 2019-04-21:
You likely won't find the help you're looking for at this page.
I suggest you try places like the Facebook groups for Marlin or Ender3 or Creality printers. Or the Reprap forums or the Toms3D forum. It's very likely that someone is already running the setup you want.

qavro, 2019-04-21:
Thank you. i wrote here because of the thread topic. Now i am running MKS Gen-L + TMC2130 +SDcard. Installing the sd card module was not that hard actually :)

wbarber69, 2019-04-21:
It shouldn’t be.  No one here is running just the sd card but if you use software spi then you should have full access to the exp ports for other add ons.