teemuatlut/TMC2130Stepper #36: StallGuard - 'PORTF' was not declared in this scope

Geekoid85, 2018-06-07:
[image]
First, thanks for your job, it amazing to control the tmc2130 as simple as that but I have a small issues. When I compile the example sketch StallGuard the debug said PORTF is not declared.
The library **work just fine** on the **"simple" sketch** and is apparently correctly installed.
Any on can help me to understand my error ?
Thanks for your time :+1: 
[image]

Geekoid85, 2018-06-07:
Actually it's an arduino UNO, but in the future an esp32.
On the official arduino documentation I only found PORTB or PORTD. What about PORTF ?
That's a new technique for me, sorry for there approximation
https://www.arduino.cc/en/Reference/PortManipulation

teemuatlut, 2018-06-07:
The example is written for Arduino Mega.
PORTF, bit 0 corresponds to D54/A0 as shown in the pinout diagram.
[image]

Geekoid85, 2018-06-07:
Ok thank you, I understand. There ports is needed for the enable, step, dir and cs pins ? The spi pins or both ?

teemuatlut, 2018-06-07:
The step port is the only one that you need to worry about as it is the only one that needs to be driven quickly.

daniel-frenkel, 2018-07-19:
> Actually it's an arduino UNO, but in the future an esp32.
> On the official arduino documentation I only found PORTB or PORTD. What about PORTF ?
> That's a new technique for me, sorry for there approximation
> https://www.arduino.cc/en/Reference/PortManipulation

@Geekoid85 Have you figured out how to use this for ESP32 or UNO? I am very confused about this port manipulation and curious if you know how to do it now. Thanks!

teemuatlut, 2018-07-19:
You can replace it with the familiar `digitalWrite`. It's not as fast but easier to use.

daniel-frenkel, 2018-07-19:
@teemuatlut I really appreciate your help and support.

If using a Nano, how can I change the following section to use digitalWrite:

```
ISR(TIMER1_COMPA_vect){
  PORTF |= 1 << 0;
  PORTF &= ~(1 << 0);
}
```

teemuatlut, 2018-07-19:
```
ISR(TIMER1_COMPA_vect){
  digitalWrite(STEP_PIN, HIGH);
  digitalWrite(STEP_PIN, LOW);
}
```

daniel-frenkel, 2018-07-19:
@teemuatlut 

Thank you so much! I made the change, but it was still not working, so I defined the SPI pins and now it works!

```
byte MOSI_PIN = 11; //SDI/MOSI (ICSP: 4, Uno: 11, Mega: 51)
byte MISO_PIN = 12; //SDO/MISO (ICSP: 1, Uno: 12, Mega: 50)
byte SCK_PIN  = 13; //CLK/SCK  (ICSP: 3, Uno: 13, Mega: 52)
```