teemuatlut/TMC2130Stepper #37: Problems executing stallguard on esp32

PrathameshJ, 2018-06-07:
I'm trying to execute stall guard example on esp32, the simple example is running fine but there are errors with stallguard. Please help

`Arduino: 1.8.5 (Windows 10), Board: "MH ET LIVE ESP32MiniKit, 80MHz, Default, 921600, None"

WARNING: library TMC2130Stepper claims to run on (avr, sam) architecture(s) and may be incompatible with your current board which runs on (esp32) architecture(s).
StallGuard:85: error: expected constructor, destructor, or type conversion before '(' token

 ISR(TIMER1_COMPA_vect){

     ^

C:\Users\Prathamesh\Documents\Arduino\libraries\TMC2130Stepper\examples\StallGuard\StallGuard.ino: In function 'void setup()':

StallGuard:66: error: 'TCCR1A' was not declared in this scope

     TCCR1A = 0;// set entire TCCR1A register to 0

     ^

StallGuard:67: error: 'TCCR1B' was not declared in this scope

     TCCR1B = 0;// same for TCCR1B

     ^

StallGuard:68: error: 'TCNT1' was not declared in this scope

     TCNT1  = 0;//initialize counter value to 0

     ^

StallGuard:69: error: 'OCR1A' was not declared in this scope

     OCR1A = 256;// = (16*10^6) / (1*1024) - 1 (must be <65536)

     ^

StallGuard:71: error: 'WGM12' was not declared in this scope

     TCCR1B |= (1 << WGM12);

                     ^

StallGuard:73: error: 'CS11' was not declared in this scope

     TCCR1B |= (1 << CS11);// | (1 << CS10);  

                     ^

StallGuard:75: error: 'TIMSK1' was not declared in this scope

     TIMSK1 |= (1 << OCIE1A);

     ^

StallGuard:75: error: 'OCIE1A' was not declared in this scope

     TIMSK1 |= (1 << OCIE1A);

                     ^

C:\Users\Prathamesh\Documents\Arduino\libraries\TMC2130Stepper\examples\StallGuard\StallGuard.ino: At global scope:

StallGuard:85: error: expected constructor, destructor, or type conversion before '(' token

 ISR(TIMER1_COMPA_vect){

    ^

C:\Users\Prathamesh\Documents\Arduino\libraries\TMC2130Stepper\examples\StallGuard\StallGuard.ino: In function 'void loop()':

StallGuard:97: error: 'TIMSK1' was not declared in this scope

     if (read_byte == '0')      { TIMSK1 &= ~(1 << OCIE1A); digitalWrite( EN_PIN, HIGH ); }

                                  ^

StallGuard:97: error: 'OCIE1A' was not declared in this scope

     if (read_byte == '0')      { TIMSK1 &= ~(1 << OCIE1A); digitalWrite( EN_PIN, HIGH ); }

                                                   ^

StallGuard:98: error: 'TIMSK1' was not declared in this scope

     else if (read_byte == '1') { TIMSK1 |=  (1 << OCIE1A); digitalWrite( EN_PIN,  LOW ); }

                                  ^

StallGuard:98: error: 'OCIE1A' was not declared in this scope

     else if (read_byte == '1') { TIMSK1 |=  (1 << OCIE1A); digitalWrite( EN_PIN,  LOW ); }

                                                   ^

StallGuard:99: error: 'OCR1A' was not declared in this scope

     else if (read_byte == '+') if (OCR1A > MAX_SPEED) OCR1A -= 20;

                                    ^

exit status 1
expected constructor, destructor, or type conversion before '(' token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

teemuatlut, 2018-06-07:
You need to figure out how to set up interrupts and timers for your STM32 board.
The macros and register names that give you the error, are AVR only.

First learn how interrupts and timers work on AVR so you know what's going on, and then adapt what you learned to STM32.

closed 2018-06-28 by PrathameshJ

DropbearNinja, 2019-11-21:
I know this was closed long ago, but hoping that @prathameshjakka found the solution as I have this same problem and fixing myself has been difficult...

prathameshjakka, 2019-11-27:
Yes I kind of did execute it but it was unreliable

technoo10201, 2020-08-06:
Hi @DropbearNinja @prathameshjakka I'm interested too regarding ESP32 & TMC2209, any code to share ? Thanks in advance !