teemuatlut/TMC2130Stepper #39: Unable to compile with PlatformIO

chippey, 2018-06-21:
Getting this string of errors:

```
.piolibdeps/TMC2130Stepper_ID1493/src/source/TMC2130Stepper.cpp:193:10: error: prototype for 'uint32_t T
MC2130Stepper::TPOWERDOWN()' does not match any in class 'TMC2130Stepper'
uint32_t TMC2130Stepper::TPOWERDOWN() { return TPOWERDOWN_sr; }
^
In file included from .piolibdeps/TMC2130Stepper_ID1493/src/source/TMC2130Stepper.cpp:1:0:
.piolibdeps/TMC2130Stepper_ID1493/src/TMC2130Stepper.h:91:8: error: candidates are: void TMC2130Stepper:
:TPOWERDOWN(uint8_t)
void TPOWERDOWN(     uint8_t input);
^
.piolibdeps/TMC2130Stepper_ID1493/src/TMC2130Stepper.h:90:11: error:                 uint8_t TMC2130Step
per::TPOWERDOWN()
uint8_t TPOWERDOWN();
^
.piolibdeps/TMC2130Stepper_ID1493/src/source/TMC2130Stepper.cpp:194:6: error: prototype for 'void TMC213
0Stepper::TPOWERDOWN(uint32_t)' does not match any in class 'TMC2130Stepper'
void TMC2130Stepper::TPOWERDOWN(uint32_t input) {
^
In file included from .piolibdeps/TMC2130Stepper_ID1493/src/source/TMC2130Stepper.cpp:1:0:
.piolibdeps/TMC2130Stepper_ID1493/src/TMC2130Stepper.h:91:8: error: candidates are: void TMC2130Stepper:
:TPOWERDOWN(uint8_t)
void TPOWERDOWN(     uint8_t input);
^
.piolibdeps/TMC2130Stepper_ID1493/src/TMC2130Stepper.h:90:11: error:                 uint8_t TMC2130Step
per::TPOWERDOWN()
uint8_t TPOWERDOWN();
^
.piolibdeps/TMC2130Stepper_ID1493/src/source/TMC2130Stepper_CHOPCONF.cpp:13:6: error: prototype for 'voi
d TMC2130Stepper::hend(int8_t)' does not match any in class 'TMC2130Stepper'
void TMC2130Stepper::hend(  int8_t  B ) { TMC_MOD_REG(CHOPCONF, HEND);  }
^
In file included from .piolibdeps/TMC2130Stepper_ID1493/src/source/TMC2130Stepper_CHOPCONF.cpp:1:0:

.piolibdeps/TMC2130Stepper_ID1493/src/TMC2130Stepper.h:140:12: error: candidates are: uint8_t TMC2130Ste
pper::hend()
uint8_t  hend();
^
.piolibdeps/TMC2130Stepper_ID1493/src/TMC2130Stepper.h:124:8: error:                 void TMC2130Stepper
::hend(uint8_t)
void hend(        uint8_t  B);
^
*** [.pioenvs/pro16MHzatmega328/lib26f/TMC2130Stepper_ID1493/source/TMC2130Stepper.cpp.o] Error 1
.piolibdeps/TMC2130Stepper_ID1493/src/source/TMC2130Stepper_CHOPCONF.cpp:30:9: error: prototype for 'int
8_t TMC2130Stepper::hend()' does not match any in class 'TMC2130Stepper'
int8_t  TMC2130Stepper::hend()  { TMC_GET_BYTE(CHOPCONF, HEND);  }
^
In file included from .piolibdeps/TMC2130Stepper_ID1493/src/source/TMC2130Stepper_CHOPCONF.cpp:1:0:
.piolibdeps/TMC2130Stepper_ID1493/src/TMC2130Stepper.h:140:12: error: candidates are: uint8_t TMC2130Ste
pper::hend()
uint8_t  hend();
^
.piolibdeps/TMC2130Stepper_ID1493/src/TMC2130Stepper.h:124:8: error:                 void TMC2130Stepper
::hend(uint8_t)
void hend(        uint8_t  B);
^
*** [.pioenvs/pro16MHzatmega328/lib26f/TMC2130Stepper_ID1493/source/TMC2130Stepper_CHOPCONF.cpp.o] Error
 1
```

teemuatlut, 2018-06-21:
Fixed now but it may take a day or two for PlatformIO to notice a new version.

chippey, 2018-06-21:
Great!

closed 2018-06-21 by chippey