teemuatlut/TMC2130Stepper #40: TPOWERDOWN compile error

thinkyhead, 2018-06-21:
```
/usr/local/share/arduino/libraries/TMC2130Stepper/src/source/TMC2130Stepper.cpp:193:10: error: prototype for 'uint32_t TMC2130Stepper::TPOWERDOWN()' does not match any in class 'TMC2130Stepper'
 uint32_t TMC2130Stepper::TPOWERDOWN() { return TPOWERDOWN_sr; }
```

teemuatlut, 2018-06-21:
#39

thinkyhead, 2018-06-21:
Although `TPWMTHRS` is fixed, `TPOWERDOWN` still isn't.

thinkyhead, 2018-06-21:
Check `TPWMTHRS` and `TPOWERDOWN` again.

#### TMC2130Stepper.h
```cpp
		// TPOWERDOWN
		uint8_t TPOWERDOWN();
		void TPOWERDOWN(					uint8_t input);
		// TSTEP
		uint32_t TSTEP();
		// TPWMTHRS
		uint32_t TPWMTHRS();
		void TPWMTHRS(						uint32_t input);
```

#### TMC2130Stepper.cpp
```cpp
uint32_t TMC2130Stepper::TPOWERDOWN() { return TPOWERDOWN_sr; }
uint8_t TMC2130Stepper::TPWMTHRS() { return TPWMTHRS_sr; }
```

thinkyhead, 2018-06-21:
The shadow registers are still all 32-bit also.

thinkyhead, 2018-06-21:
I don't know which is supposed to be changed to `uint8_t` or I'd submit a PR.

p3p, 2018-06-21:
> #39 

with 2.4.1 there is still an issue, the types don't match.
```c++
		// TPOWERDOWN
		uint8_t TPOWERDOWN();
		void TPOWERDOWN(					uint8_t input);
		// TSTEP
		uint32_t TSTEP();
		// TPWMTHRS
		uint32_t TPWMTHRS();
		void TPWMTHRS(	
```

```c++
// W: TPOWERDOWN
uint32_t TMC2130Stepper::TPOWERDOWN() { return TPOWERDOWN_sr; }
void TMC2130Stepper::TPOWERDOWN(uint32_t input) {
	TPOWERDOWN_sr = input;
	TMC_WRITE_REG(TPOWERDOWN);
}
///////////////////////////////////////////////////////////////////////////////////////
// R: TSTEP
uint32_t TMC2130Stepper::TSTEP() { TMC_READ_REG_R(TSTEP); }
///////////////////////////////////////////////////////////////////////////////////////
// W: TPWMTHRS
uint8_t TMC2130Stepper::TPWMTHRS() { return TPWMTHRS_sr; }
void TMC2130Stepper::TPWMTHRS(uint8_t input) {
	TPWMTHRS_sr = input;
	TMC_WRITE_REG(TPWMTHRS);
}
```

teemuatlut, 2018-06-21:
This is what I get for doing this late at night without testing and then forgetting that PIO doesn't work with tags...

Should be fixed now in 2.4.2. I ran it with both PIO and A IDE.
None of these changes were supposed to be a real release yet and no tags were made.

teemuatlut, 2018-06-21:
> The shadow registers are still all uint32_t also.

That's a small change I can do but I don't think I'll be making too many micro-optimizations as we will not be using this library for much longer.

closed 2018-06-22 by thinkyhead