teemuatlut/TMC2130Stepper #41: Use of TMC drivers in SPI cascade mode.

pgScorpio, 2018-06-22:
The current SPI support for the trinamic drivers requires all tmc chips to be connected to miso and mosi of the processor and a separate chip select pin for each driver.

For many (3D printer) configurations however this is a pain in the ass because there are not enough pins available/accessible (many times forcing you to use the endstop-switch pins for cs and use tmc stallguard as endstop, which has its own problems).

The TMC chips however can be used in cascade where miso/mosi are connected from chip to chip and they all share one common chip select. This requires only 3 connections to the processor to control all connected TMC chips and this would also simplify the interconnection of the stepsticks tremendously !

[TMC-SPI.pdf](https://github.com/teemuatlut/TMC2130Stepper/files/2127483/TMC-SPI.pdf)

teemuatlut, 2018-06-22:
I might entertain this idea with the TMCStepper library that will be the main focus going forward.
It might not even be too difficult to implement.

Grogyan, 2018-06-25:
I propose using a Serial Shift register, such the 74HC595 or 74HC164 to control the CS lines

pgScorpio, 2018-06-26:
@grogyan
A: Please explain how to use a serial shift register without using extra processor pins....
B: Why use a proprietary method while the device supports standard cascade mode ?

Grogyan, 2018-06-26:
The TMC2130 does not have a cascadable feature, as far as I am aware
The TMC5130 and 5160 do however have a "cascadable" feature and support a 1-wire bus.

However, using a serial shift register, is an easy method to support any SPI system to control the Chip Select pins, and both the 74HC595 and 74HC164 can be cascaded.
The 74HC164 uses one less pin than the 595, however, I prefer a separate pin to shift all the data at once.

74HC164 needs
Clock, Data, and maybe a Reset pin
You can potentially use the same SPI clock and SPI data MOSI pins as the TMC drivers, leaving one additional pin for the latching of the CS lines from the shift register.

As an additional bonus to using a shift register, you can use the same hardware SPI bus and shift register to control other SPI enables devices, for example the MAX31855 thermocouple, which I use predominantly over thermistors on a hot end.

pgScorpio, 2018-06-26:
See chapter 4.2 of the tmc2130 datasheet:
"If  more  than  40  clocks  are  driven,  the  additional  bits  shifted  into  SDI  are  shifted  out  on  SDO  after  a 40-clock delay  through  an internal shift register.  This  can be used for  daisy  chaining  multiple chips."

Concerning the use of shift registers:
Yes I know you can use shift registers as spi "io expanders", but that does not completely solve the problem of already to few pins available and it will certainly not simplify the interconnection of the tmc chips to the processor (adding an extra chip + connections)..... also I think this would have more impact on the software...

Grogyan, 2018-06-26:
@pgScorpio I stand corrected 👍 
I somehow missed that part of the datasheet like 10 times. Lol