teemuatlut/TMC2130Stepper #42: v2.4.2 and Marlin Bugfix-2.0 Issue

MikeLud, 2018-06-27:
@teemuatlut 
I am having issues with v2.4.2 and Marlin Bugfix-2.0 (6/12/18) after loading the firmware I get the below errors. If I set the TMC2130 Library to version a80fb0274e7e6e66c24c4ad3b34dd87b92efd0ca everything works with no errors. Please let me know if you need additional info and/or if I should also post the issue with Marlin

 Connecting...
start
Printer is now online.
echo:Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: (none, default config)
echo:Compiled: Jun 26 2018
echo: Free Memory: 2105  PlannerBufferBytes: 1664
echo:V55 stored settings retrieved (815 bytes; crc 365)
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X100.34 Y100.34 Z400.00 E100.00
echo:Maximum feedrates (units/s):
echo:  M203 X400.00 Y400.00 Z8.00 E50.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X400 Y400 Z100 E1000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P400.00 R1000.00 T400.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev> E<max_e_jerk>
echo:  M205 B20000 S0.00 T0.00 J0.02
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Auto Bed Leveling:
echo:  M420 S0 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P26.06 I1.80 D94.47
echo:Z-Probe Offset (mm): (mm):
echo:  M851 Z-1.60
echo:Stepper driver current:
echo:  M906 X575 Y575 Z740
  M906 T0 E740
echo:Linear Advance:
echo:  M900 K0.00
X driver error detected:
overtemperature
short to ground (coil A)
short to ground (coil B)
		X	Y	Z	E0
Enabled		false	false	false	false
Set current	575	575	740	740
RMS current	550	994	734	1325
MAX current	776	1402	1035	1868
Run current	17/31	17/31	23/31	23/31
Hold current	8/31	8/31	11/31	11/31
CS actual		31/31	0/31	31/31	0/31
PWM scale		255	0	255	0
vsense		1=.18	0=.325	1=.18	0=.325
stealthChop	true	false	true	false
msteps		0	256	0	256
tstep		4294967295	0	4294967295	0
pwm
threshold		0	0	0	0
[mm/s]		-	-	-	-
OT prewarn	true	false	true	false
OT prewarn has
been triggered	false	false	false	false
off time		15	0	15	0
blank time	54	16	54	16
hysteresis
-end		12	-3	12	-3
-start		8	1	8	1
Stallguard thrs	0	0	0	0
DRVSTATUS	X	Y	Z	E0
stallguard		X		X
sg_result		1023	0	1023	0
fsactive		X		X
stst		X		X
olb		X		X
ola		X		X
s2gb		X		X
s2ga		X		X
otpw		X		X
ot		X		X
Driver registers:	X = 0xFF:FF:FF:FF
	Y = 0x00:00:00:00
	Z = 0xFF:FF:FF:FF
	E0 = 0x00:00:00:00
Error:Printer halted. kill() called!
Error:Printer halted. kill() called!
[image]

teemuatlut, 2018-06-27:
The 2.4.x wasn't supposed to be a release and there is not release tag present for Arduino IDE. The PIO platform just works a bit different (worse). You can keep using the previous versions.
What's the motherboard you're using?

MikeLud, 2018-06-27:
@teemuatlut 
Thanks for the response, I will continue use the previous version. I have a MKS Gen L V1.0 and have the TMC2130 wired as in the below link. One note, I am still using end stop.
[MKS_Gen-L_TMC2130_SPI_Sensor-less_Homing_Wiring_Diagram (1).pdf](https://github.com/teemuatlut/TMC2130Stepper/files/2139805/MKS_Gen-L_TMC2130_SPI_Sensor-less_Homing_Wiring_Diagram.1.pdf)

MikeLud, 2018-06-27:
@teemuatlut 
One suggestion, you might want to specify the current release version in Marlin's platformio.ini file as I did see below line 34, before it just had "TMC2130Stepper" for line 34

[image]

teemuatlut, 2018-06-27:
The downside is that if I fix something in the library I also have to do a pull request to Marlin before the changes actually take effect.

MoellerDi, 2018-06-30:
Same issue here; MKS Gen 1.4 with TMC2130 on XYZE. Setting an older library version (e.g.2.3.0) in Marlin's platformio.ini file helps to get it fixed for me

MikeLud, 2018-06-30:
@MoellerDi 
That is what I did and everything is back to normal, thanks

teemuatlut, 2018-06-30:
I've reverted the communication change for 2.4.3 which will be out whenever the PIO crawler finds it.
Arduino IDE still unaffected because no releases were made.
I'm also keeping this open a bit longer.