teemuatlut/TMC2130Stepper #45: ESP32 Compatibilty

daniel-frenkel, 2018-07-20:
First, thank you for spending the time to create this library. I appreciate it very much.

When compiling for an ESP32 device, I receive the following errors. I am assuming the library is not compatible with ESP32 correct?

```
WARNING: library TMC2130Stepper claims to run on (avr, sam) architecture(s) and may be incompatible with your current board which runs on (esp32) architecture(s).

StallGuardESP32:90: error: expected constructor, destructor, or type conversion before '(' token

 ISR(TIMER1_COMPA_vect){

     ^

In function 'void setup()':

StallGuardESP32:71: error: 'TCCR1A' was not declared in this scope

     TCCR1A = 0;// set entire TCCR1A register to 0

     ^

StallGuardESP32:72: error: 'TCCR1B' was not declared in this scope

     TCCR1B = 0;// same for TCCR1B

     ^

StallGuardESP32:73: error: 'TCNT1' was not declared in this scope

     TCNT1  = 0;//initialize counter value to 0

     ^

StallGuardESP32:74: error: 'OCR1A' was not declared in this scope

     OCR1A = 60;// = (16*10^6) / (1*1024) - 1 (must be <65536)

     ^

StallGuardESP32:76: error: 'WGM12' was not declared in this scope

     TCCR1B |= (1 << WGM12);

                     ^

StallGuardESP32:78: error: 'CS11' was not declared in this scope

     TCCR1B |= (1 << CS11);// | (1 << CS10);  

                     ^

StallGuardESP32:80: error: 'TIMSK1' was not declared in this scope

     TIMSK1 |= (1 << OCIE1A);

     ^

StallGuardESP32:80: error: 'OCIE1A' was not declared in this scope

     TIMSK1 |= (1 << OCIE1A);

                     ^

StallGuardESP32:90: error: expected constructor, destructor, or type conversion before '(' token

 ISR(TIMER1_COMPA_vect){

    ^

StallGuardESP32:102: error: 'TIMSK1' was not declared in this scope

     if (read_byte == '0')      { TIMSK1 &= ~(1 << OCIE1A); digitalWrite( EN_PIN, HIGH ); }

                                  ^

StallGuardESP32:102: error: 'OCIE1A' was not declared in this scope

     if (read_byte == '0')      { TIMSK1 &= ~(1 << OCIE1A); digitalWrite( EN_PIN, HIGH ); }

                                                   ^

StallGuardESP32:103: error: 'TIMSK1' was not declared in this scope

     else if (read_byte == '1') { TIMSK1 |=  (1 << OCIE1A); digitalWrite( EN_PIN,  LOW ); }

                                  ^

StallGuardESP32:103: error: 'OCIE1A' was not declared in this scope

     else if (read_byte == '1') { TIMSK1 |=  (1 << OCIE1A); digitalWrite( EN_PIN,  LOW ); }

                                                   ^

StallGuardESP32:104: error: 'OCR1A' was not declared in this scope

     else if (read_byte == '+') if (OCR1A > MAX_SPEED) OCR1A -= 20;

                                    ^

StallGuardESP32:104: error: suggest explicit braces to avoid ambiguous 'else' [-Werror=parentheses]

     else if (read_byte == '+') if (OCR1A > MAX_SPEED) OCR1A -= 20;

             ^

cc1plus.exe: some warnings being treated as errors

Using library SPI at version 1.0
Using library TMC2130Stepper at version 2.3.0

exit status 1

expected constructor, destructor, or type conversion before '(' token
```

PrathameshJ, 2018-07-20:
yes, I'm too trying to make the library work with esp32, this library works with avr, sam architecture.
I'm still trying to understand how to use timers and interrupt on the esp32.

teemuatlut, 2018-07-24:
The registers are AVR specific. You need to use ESP register definitions to achieve the same for the code to work.

franzuu, 2018-07-25:
Same here, would also really need help to make this library work on esp32 as I'm not a great developer. I noticed the same type of issue has already been closed for this library, but I think most people around are not very into modifing a library on their own (or whatever needs to be changed to get it working). It would be very helpful if more instructions and detailed guidance would be made available to everyone or some more experienced developer maybe has time to make it work on esp32 :).

daniel-frenkel, 2018-07-25:
I have it working with ESP32 now. I will post the code later today. The way to make it work is to use the DIAG1 pin on the 2130, which goes LOW with stallGuard is sensed. 

You then need to remove all registers from the code, just delete them, and delete the PORTF code as well. 

Also set Active High to 0.

That's all I remember until I can look at the code

myseg, 2018-07-26:
I believe this defeats the purpose, as the code's main function is to plot the load, not just to flag a stall state. I am also very much interested to make it work with the esp32, I am currently digging through the espressif database to find some specific interrupt, port and timer names for it... After that, the only thing is to replace them in the code manually.

myseg, 2018-07-26:
So I just realized that all the avr related timer code is there to just move the stepper and control the speed without using the delay() functions... I will write a replacement for it when i get home on the weekend.

teemuatlut, 2018-07-27:
You're welcome to try but from experience the delay approach didn't work. The driver/motor stops for a brief duration anytime it reads the driver status and so the reported value isn't valid.

myseg, 2018-07-28:
So after ton of searching and learning, I have found out that the esp32 has a quite programmer friendly timer interrupt routine, no need to look up for registries or such, just simple functions.
Here is the code I modified myself (aka: most of it is copied from other github posts)
[StallGuard_ESP32.zip](https://github.com/teemuatlut/TMC2130Stepper/files/2238309/StallGuard_ESP32.zip)

Tested with a smalll nema11 motor, works as it should, however the load graph is quite noisy, don't know if that is normal. Set the check interval to .01 sec, and I can see a somewhat clear sinusoidal line, it might be that my motor is old and stiff at some points.

myseg, 2018-08-02:
Keep in mind that the stall value is a very sensitive number, changes from motor to motor, current to current, and very slight changes make all the difference. In my case, the value of 9 instantly raised the stall flag (graph = 0), and 12 would never stall the motor even after several skipped steps.
If the motor spins, and you can increase and decrease the speed, the code is working. The problem must be with the value, or wiring, or not setting the pin numbers correctly, or not having an SPI-configured driver.
(also, set the serial monitor to newline only)

edwardocano, 2020-04-29:
Hello! @teemuatlut do you know if i move the steppers with accelstepper library i could get correct values of driver_1.SG_RESULT()? 
I have been trying but i just get a square wave with zeros like lowest values.