teemuatlut/TMC2130Stepper #46: Einsy Rambo

Djamuka, 2018-07-31:
Hi, I have an einsy rambo and the spreadCycle won't work with marlin and sensorless homming, I try a lots of thing but motor don't move in this mode.
So I try the spreadcycle example to see if it's marlin or the lib, and the same thing happen motor don't move in this mode.
I connect to X and changed the pin by this:
#define EN_PIN       29  
#define DIR_PIN      49  
#define STEP_PIN     37  
#define CS_PIN       41  
#define STEP_PORT PORTC  // Register to match with STEP_PIN
#define STEP_BIT      0

Board from Prusa Shop, and the motor work fine with simple example, so it's not a bad connection or pin assignement.

Can you help me please?

teemuatlut, 2018-07-31:
`Calibrate spreadCycle` example doesn't attempt to move the motor (much). Try the stallGuard example.
I used your pin definitions and the stallGuard example _works as intended_.

Apologies if the unnecessary port definitions were misleading. I'm removing them on the next release.

Djamuka, 2018-07-31:
Thanks for the answer, I try the Stallguard example and nothing happen, Try different value for the stall, but nothing.
Motor is blocking, and didn't move.

teemuatlut, 2018-07-31:
Does the motor lock up?
What's the serial output?

Djamuka, 2018-07-31:
The motor do just one step and block i can't turn him..

the serial output:
0 35 305
0 34 305
0 34 305
0 38 305
0 34 305
0 39 305
0 35 305
0 35 305
0 36 305
0 38 305
0 33 305
0 38 305
0 32 305
0 38 305
0 33 305
0 35 305
0 33 305
0 34 305
0 39 305
0 39 305
0 39 305
0 40 305
0 36 305
0 38 305
0 35 305
0 36 305
0 35 305
....

teemuatlut, 2018-07-31:
Your SG results and your current seem a bit low, but it reminds me of something.
The example is written with standard RAMPS in mind and the UltiMachine boards use slightly different hardware components.

Please do the following changes:
```cpp
Add:
#define R_SENSE 0.22  // Match sense resistor value to your board/driver
```
```cpp
Change from
driver.rms_current(600); // mA
to
driver.rms_current(600, 1.0, R_SENSE); // mA
```
and
```cpp
from
Serial.println(rms_current((drv_status & CS_ACTUAL_bm)>>CS_ACTUAL_bp), DEC);
to
Serial.println(rms_current((drv_status & CS_ACTUAL_bm)>>CS_ACTUAL_bp, R_SENSE), DEC);
```

Djamuka, 2018-07-31:
with the change the motor is always blocked, and with more force...
FYI I don'' think if it's revelant but my motor is a SY42STH47 with rated current at 1.68A...

Here the serial output:
Start...
0 63 598
0 56 598
0 63 598
0 61 598
0 63 598
0 63 598
0 63 598
0 62 598
0 63 598
0 63 598
0 61 598
0 59 598
0 62 598
0 63 598
0 65 598
0 62 598
0 58 598
0 63 598
0 63 598
0 63 598
0 61 598

teemuatlut, 2018-07-31:
No surprise it didn't fix the motor moving. It should spin even with a wrong current setting.
Try this next:
```cpp
ISR(TIMER1_COMPA_vect){
  digitalWrite(STEP_PIN, HIGH);
  digitalWrite(STEP_PIN, LOW);
}
```

teemuatlut, 2018-07-31:
Does the motor work in Marlin with stealthChop?
Does the motor work with the Prusa firmware?

I'm thinking that for whatever reason PC0 does not correspond to X_STEP_PIN on your board.
But even if you had mixed Einsy Rambo with Einsy Retro (not saying you have), the pins are still the same.
Could you maybe attach a picture of your board just to make sure?

Also just to make sure, have you installed the Rambo board addon from UltiMachine?

Djamuka, 2018-07-31:
Yes Motor work in marlin stealthChop.
I Will try the prusa FW now...

PC0 not connect to X_Step_Pin...very strange it's the same board prusa use and the X_Step_Pin is the Same 37, PC0...

For the board it's einsy rambo 1.1 and yes the Addon is installed:
https://nsa39.casimages.com/img/2018/07/31/180731103956353999.jpg
https://nsa39.casimages.com/img/2018/07/31/180731104022610868.jpg

teemuatlut, 2018-07-31:
Well then as a thing I'd tell you to try the 2.0.x branch.
I'll try running Marlin on my board tomorrow. Maybe there's something wrong with the current coding or something.
Overall your problem was that the motor doesn't even move if configured to spreadCycle? Disregarding sensorless homing for now, first try to get both modes working.

Djamuka, 2018-07-31:
With Prusa FW motor in X "sensorless homing" perfectly, but if the belt is not connected...when belt is in place motor make noise and don't move, there  are not the same as prusa so the current is not set for mine...
sensorless homing work if I stop the shaft with finger... 
For problem if the sensorless homing didn't work I buy a board with some addon that can't be used...it's not nice!
By the way thanks for the time, and hope you see something when yru try runing Marlin on your einsy.

Djamuka, 2018-07-31:
it's good news  it work with marlin, bad news for me, i'm the error!!
I just set the rms_current at 1000, the max is 960...so when I set the driver to 960 it work like a charm...I didn't think there was a protect for this.....
Apologies and thanks again!

Djamuka, 2018-08-01:
Exactly it's Marlin 1.1.8, it work with 960 motor move with enought strenght so not a problem now, I can finish my custom 3D Printer ;)
For the 960 maybe an error message like the other when you don't set TMC2130 on Einsy, but if I'm the only one that made this error, it's useless to have this report.
For the Rsense it was set to 0.22 in the first flash just after I read the Wiki,the max motor current is write too but just wanted to push a lilttle bit the motor current...
Thanks a lot for the lib and your time.
have a nice day.

closed 2018-08-07 by teemuatlut