teemuatlut/TMC2130Stepper #47: Can't compile in Arduino IDE, "Update TMC2130Stepper library to 2.2.1 or newer."

Flawl3ssSWE, 2018-08-05:
**I configured my Marlin for my new TMC2130 v1.1 today and when I hit the verify button I received this error:**

`
In file included from sketch\MarlinConfig.h:39:0,

                 from sketch\G26_Mesh_Validation_Tool.cpp:27:

SanityCheck.h:1439: error: #error "Update TMC2130Stepper library to 2.2.1 or newer."

     #error "Update TMC2130Stepper library to 2.2.1 or newer."

      ^

exit status 1
#error "Update TMC2130Stepper library to 2.2.1 or newer."
`
**I have installed the library from Arduino 2.3.0 and tried downloading it from GitHub, I also tried version 2.2.1 but still no luck. I also tried different Arduino IDE versions but still no luck. Have anyone got any idea how to fix it.

If you need the TMC2130 part from the configuration_adv.h here it is:**
// @section tmc_smart

/**
 * Enable this for SilentStepStick Trinamic TMC2130 SPI-configurable stepper drivers.
 *
 * You'll also need the TMC2130Stepper Arduino library
 * (https://github.com/teemuatlut/TMC2130Stepper).
 *
 * To use TMC2130 stepper drivers in SPI mode connect your SPI pins to
 * the hardware SPI interface on your board and define the required CS pins
 * in your `pins_MYBOARD.h` file. (e.g., RAMPS 1.4 uses AUX3 pins `X_CS_PIN 53`, `Y_CS_PIN 49`, etc.).
 * You may also use software SPI if you wish to use general purpose IO pins.
 */
#define HAVE_TMC2130
#if ENABLED(HAVE_TMC2130)  // Choose your axes here. This is mandatory!
  #define X_IS_TMC2130
  //#define X2_IS_TMC2130
  #define Y_IS_TMC2130
  //#define Y2_IS_TMC2130
  #define Z_IS_TMC2130
  //#define Z2_IS_TMC2130
  #define E0_IS_TMC2130
  #define E1_IS_TMC2130
  //#define E2_IS_TMC2130
  //#define E3_IS_TMC2130
  //#define E4_IS_TMC2130
#endif

/**
 * Enable this for SilentStepStick Trinamic TMC2208 UART-configurable stepper drivers.
 * Connect #_SERIAL_TX_PIN to the driver side PDN_UART pin with a 1K resistor.
 * To use the reading capabilities, also connect #_SERIAL_RX_PIN
 * to PDN_UART without a resistor.
 * The drivers can also be used with hardware serial.
 *
 * You'll also need the TMC2208Stepper Arduino library
 * (https://github.com/teemuatlut/TMC2208Stepper).
 */
//#define HAVE_TMC2208
#if ENABLED(HAVE_TMC2208)  // Choose your axes here. This is mandatory!
  //#define X_IS_TMC2208
  //#define X2_IS_TMC2208
  //#define Y_IS_TMC2208
  //#define Y2_IS_TMC2208
  //#define Z_IS_TMC2208
  //#define Z2_IS_TMC2208
  //#define E0_IS_TMC2208
  //#define E1_IS_TMC2208
  //#define E2_IS_TMC2208
  //#define E3_IS_TMC2208
  //#define E4_IS_TMC2208
#endif

#if ENABLED(HAVE_TMC2130) || ENABLED(HAVE_TMC2208)

  #define R_SENSE           0.11  // R_sense resistor for SilentStepStick2130
  #define HOLD_MULTIPLIER    0.5  // Scales down the holding current from run current
  #define INTERPOLATE       true  // Interpolate X/Y/Z_MICROSTEPS to 256

  #define X_CURRENT          600  // rms current in mA. Multiply by 1.41 for peak current.
  #define X_MICROSTEPS        16  // 0..256

  #define Y_CURRENT          600
  #define Y_MICROSTEPS        16

  #define Z_CURRENT          700
  #define Z_MICROSTEPS        16

  #define X2_CURRENT         800
  #define X2_MICROSTEPS       16

  #define Y2_CURRENT         800
  #define Y2_MICROSTEPS       16

  #define Z2_CURRENT         800
  #define Z2_MICROSTEPS       16

  #define E0_CURRENT         600
  #define E0_MICROSTEPS       16

  #define E1_CURRENT         600
  #define E1_MICROSTEPS       16

  #define E2_CURRENT         800
  #define E2_MICROSTEPS       16

  #define E3_CURRENT         800
  #define E3_MICROSTEPS       16

  #define E4_CURRENT         800
  #define E4_MICROSTEPS       16

  /**
   * Use software SPI for TMC2130.
   * The default SW SPI pins are defined the respective pins files,
   * but you can override or define them here.
   */
  //#define TMC_USE_SW_SPI
  //#define TMC_SW_MOSI       -1
  //#define TMC_SW_MISO       -1
  //#define TMC_SW_SCK        -1

  /**
   * Use Trinamic's ultra quiet stepping mode.
   * When disabled, Marlin will use spreadCycle stepping mode.
   */
  #define STEALTHCHOP

  /**
   * Monitor Trinamic TMC2130 and TMC2208 drivers for error conditions,
   * like overtemperature and short to ground. TMC2208 requires hardware serial.
   * In the case of overtemperature Marlin can decrease the driver current until error condition clears.
   * Other detected conditions can be used to stop the current print.
   * Relevant g-codes:
   * M906 - Set or get motor current in milliamps using axis codes X, Y, Z, E. Report values if no axis codes given.
   * M911 - Report stepper driver overtemperature pre-warn condition.
   * M912 - Clear stepper driver overtemperature pre-warn condition flag.
   * M122 S0/1 - Report driver parameters (Requires TMC_DEBUG)
   */
  //#define MONITOR_DRIVER_STATUS

  #if ENABLED(MONITOR_DRIVER_STATUS)
    #define CURRENT_STEP_DOWN     50  // [mA]
    #define REPORT_CURRENT_CHANGE
    #define STOP_ON_ERROR
  #endif

  /**
   * The driver will switch to spreadCycle when stepper speed is over HYBRID_THRESHOLD.
   * This mode allows for faster movements at the expense of higher noise levels.
   * STEALTHCHOP needs to be enabled.
   * M913 X/Y/Z/E to live tune the setting
   */
  //#define HYBRID_THRESHOLD

  #define X_HYBRID_THRESHOLD     100  // [mm/s]
  #define X2_HYBRID_THRESHOLD    100
  #define Y_HYBRID_THRESHOLD     100
  #define Y2_HYBRID_THRESHOLD    100
  #define Z_HYBRID_THRESHOLD       3
  #define Z2_HYBRID_THRESHOLD      3
  #define E0_HYBRID_THRESHOLD     30
  #define E1_HYBRID_THRESHOLD     30
  #define E2_HYBRID_THRESHOLD     30
  #define E3_HYBRID_THRESHOLD     30
  #define E4_HYBRID_THRESHOLD     30

  /**
   * Use stallGuard2 to sense an obstacle and trigger an endstop.
   * You need to place a wire from the driver's DIAG1 pin to the X/Y endstop pin.
   * X, Y, and Z homing will always be done in spreadCycle mode.
   *
   * X/Y/Z_HOMING_SENSITIVITY is used for tuning the trigger sensitivity.
   * Higher values make the system LESS sensitive.
   * Lower value make the system MORE sensitive.
   * Too low values can lead to false positives, while too high values will collide the axis without triggering.
   * It is advised to set X/Y/Z_HOME_BUMP_MM to 0.
   * M914 X/Y/Z to live tune the setting
   */
  //#define SENSORLESS_HOMING // TMC2130 only

  #if ENABLED(SENSORLESS_HOMING)
    #define X_HOMING_SENSITIVITY  8
    #define Y_HOMING_SENSITIVITY  8
    #define Z_HOMING_SENSITIVITY  8
  #endif

  /**
   * Enable M122 debugging command for TMC stepper drivers.
   * M122 S0/1 will enable continous reporting.
   */
  #define TMC_DEBUG

  /**
   * M915 Z Axis Calibration
   *
   * - Adjust Z stepper current,
   * - Drive the Z axis to its physical maximum, and
   * - Home Z to account for the lost steps.
   *
   * Use M915 Snn to specify the current.
   * Use M925 Znn to add extra Z height to Z_MAX_POS.
   */
  //#define TMC_Z_CALIBRATION
  #if ENABLED(TMC_Z_CALIBRATION)
    #define CALIBRATION_CURRENT 250
    #define CALIBRATION_EXTRA_HEIGHT 10
  #endif

  /**
   * You can set your own advanced settings by filling in predefined functions.
   * A list of available functions can be found on the library github page
   * https://github.com/teemuatlut/TMC2130Stepper
   * https://github.com/teemuatlut/TMC2208Stepper
   *
   * Example:
   * #define TMC_ADV() { \
   *   stepperX.diag0_temp_prewarn(1); \
   *   stepperY.interpolate(0); \
   * }
   */
  #define TMC_ADV() {  }

#endif // TMC2130 || TMC2208

teemuatlut, 2018-08-05:
What Marlin branch are you using?
Please paste in the full error log produced.

Flawl3ssSWE, 2018-08-05:
I am using Jim Browns easy config: https://github.com/JimBrown/MarlinTarantula so it is bugfix 1.1.x, this is the error log I think:

Build options changed, rebuilding all
In file included from sketch\MarlinConfig.h:40:0,

                 from sketch\G26_Mesh_Validation_Tool.cpp:27:

SanityCheck.h:1439: error: #error "Update TMC2130Stepper library to 2.2.1 or newer."

     #error "Update TMC2130Stepper library to 2.2.1 or newer."

      ^

exit status 1
#error "Update TMC2130Stepper library to 2.2.1 or newer."

Flawl3ssSWE, 2018-08-05:
[image]

Flawl3ssSWE, 2018-08-05:
the current version of the TMC2130Stepper library was 2.4.3 downloaded here from GitHub, I have also tried restarting the IDE and my computer.

teemuatlut, 2018-08-06:
Try re-downloading the firmware. I was able to compile it with `TMC2130` enabled for X Y Z E0.
If that doesn't help, then attach your config files here.

Flawl3ssSWE, 2018-08-06:
Since it was not working earlier today, I decided to convert to Marlin 2.0 and now I don’t have that error anymore.

Thanks for the help!

closed 2018-08-07 by teemuatlut