teemuatlut/TMC2130Stepper #48: TMC2130 only reading  0xFF:FF:FF:FF on all drivers

Flawl3ssSWE, 2018-08-08:
Uploaded my firmware today and tried moving an axis using G1 X10 it and nothing happened, so sent M122 and it sent back
`SENDING:M122
		X	Y	Z	E0
Enabled		true	false	false	false
Set current	800	800	800	800
RMS current	795	795	795	795
MAX current	1121	1121	1121	1121
Run current	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31
CS actual		31/31	31/31	31/31	31/31
PWM scale		255	255	255	255
vsense		1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true
msteps		0	0	0	0
tstep		4294967295	4294967295	4294967295	4294967295
pwm
threshold		0	0	0	0
[mm/s]		-	-	-	-
OT prewarn	true	true	true	true
OT prewarn has
been triggered	false	false	false	false
off time		15	15	15	15
blank time	54	54	54	54
hysteresis
-end		12	12	12	12
-start		8	8	8	8
Stallguard thrs	8	8	0	0
DRVSTATUS	X	Y	Z	E0
stallguard		X	X	X	X
sg_result		1023	1023	1023	1023
fsactive		X	X	X	X
stst		X	X	X	X
olb		X	X	X	X
ola		X	X	X	X
s2gb		X	X	X	X
s2ga		X	X	X	X
otpw		X	X	X	X
ot		X	X	X	X
Driver registers:	X = 0xFF:FF:FF:FF
	Y = 0xFF:FF:FF:FF
	Z = 0xFF:FF:FF:FF
	E0 = 0xFF:FF:FF:FF`

Here are my Config files (conf.h conf_adv.h):

[Conf files.zip](https://github.com/teemuatlut/TMC2130Stepper/files/2271773/Conf.files.zip)

I put everything together using: 
[MKS_Gen-L_TMC2130_SPI_Sensor-less_Homing_Wiring_Diagram.pdf](https://github.com/teemuatlut/TMC2130Stepper/files/2271776/MKS_Gen-L_TMC2130_SPI_Sensor-less_Homing_Wiring_Diagram.pdf)

What should I do now to get it all working?

Flawl3ssSWE, 2018-08-08:
[image]
here is the pronterface response since it put it togheter to one text.

teemuatlut, 2018-08-08:
Hardware error.
Check your wiring and grimps. Change wires. Try to get one driver working first.
Try software SPI if other methods fail.

Flawl3ssSWE, 2018-08-09:
Why is not my e1 extruder working? It is reading E1 = 0xE0:0C:00:00
[image]

Flawl3ssSWE, 2018-08-10:
Got it working

closed 2018-08-10 by Flawl3ssSWE

workheart, 2018-08-10:
could you tell how? problems in the wireing?

Flawl3ssSWE, 2018-08-10:
I have fixed it, it is working now.

dmitrif, 2018-09-30:
What was the issue?

Flawl3ssSWE, 2018-09-30:
@dmitrif I just redid all of the wiring and it started working as it should.

dmitrif, 2018-10-09:
For me the issue was that I got clones, which are wired to operate in standalone mode. So had to desolder the standalone bridge, and solder the config pins.

Cheers!