teemuatlut/TMC2130Stepper #49: Feedback on m122 for other axis than xyz

workheart, 2018-08-08:
Hey everybody , I am curently gambeling arround with more TMC2130 Drivers, its only on the Bench but maybe you could follow me:
ich got TMC2130's including SPI and DIag0 on all 6axis (on my new printer this will be XYZ X2 E0 E1 XYZ are working fine, i know how to asign X2 etc thats not new, 
but on TMC2130 Debug M122 I ony get status reports on XYZ never on X2 or any of the extruders, 
If i set current via 
SENDING:M906 X I1 500 // it always reports (like it would work) back current of all the other axis, but not of the "uncommen ones"
X driver current: 800
Y driver current: 900
Z driver current: 900
This is M122 output:
SENDING:M122
		X	Y	Z
Enabled		false	false	false
Set current	800	900	900
RMS current	795	887	887
MAX current	1121	1251	1251
Run current	25/31	28/31	28/31
Hold current	12/31	14/31	14/31
CS actual		12/31	14/31	14/31
PWM scale		128	128	6
vsense		1=.18	1=.18	1=.18
stealthChop	true	true	true
msteps		16	16	16
tstep		1048575	1048575	1048575
pwm
threshold		0	0	0
[mm/s]		-	-	-
OT prewarn	false	false	false
OT prewarn has
been triggered	false	false	false
off time		5	5	5
blank time	24	24	24
hysteresis
-end		2	2	2
-start		3	3	3
Stallguard thrs	8	8	8
DRVSTATUS	X	Y	Z
stallguard
sg_result		0	0	0
fsactive
stst		X	X	X
olb		X	X
ola		X	X
s2gb
s2ga
otpw
ot
Driver registers:	X = 0xE0:0C:00:00
	Y = 0xE0:0E:00:00
	Z = 0x80:0E:00:00

could someone tell me if ther is just not feedback on this (X2 etc) or am I mistaking somwere?

btw hardware is completly custom build arround arduinomega and i checkt the pins for correct conection.

teemuatlut, 2018-08-09:
Try the Marlin bugfix branches.

workheart, 2018-08-09:
This is on marlin 2.0 bugfix from last sunday

teemuatlut, 2018-08-09:
Update the branch. There have recently been changes regarding this issue.

workheart, 2018-08-09:
okay thanks, thaths it
`SENDING:M122
		X	X2	Y	Z	E0
Enabled		false	false	false	false	false
Set current	900	800	900	900	800
RMS current	887	795	887	887	795
MAX current	1251	1121	1251	1251	1121
Run current	28/31	25/31	28/31	28/31	25/31
Hold current	14/31	12/31	14/31	14/31	12/31
CS actual		14/31	12/31	14/31	14/31	12/31
PWM scale		4	4	4	4	4
vsense		1=.18	1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true	true
msteps		16	16	16	16	16
tstep		1048575	1048575	1048575	1048575	1048575
pwm
threshold		0	0	0	0	0
[mm/s]		-	-	-	-	-
OT prewarn	false	false	false	false	false
OT prewarn has
been triggered	false	false	false	false	false
off time		5	5	5	5	5
blank time	24	24	24	24	24
hysteresis
-end		2	2	2	2	2
-start		3	3	3	3	3
Stallguard thrs	8	8	8	8	0
DRVSTATUS	X	X2	Y	Z	E0
stallguard
sg_result		0	0	0	0	0
fsactive
stst		X	X	X	X	X
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:	X = 0x80:0E:00:00
	X2 = 0x80:0C:00:00
	Y = 0x80:0E:00:00
	Z = 0x80:0E:00:00
	E0 = 0x80:0C:00:00`

closed 2018-08-09 by workheart