teemuatlut/TMC2130Stepper #50: All Steppers not moving after a few simple pins resolder on TMC2130's

bossman74, 2018-08-13:
i need help with my TMC2130, I'm not sure if this is the right place, if not, please delete.

I'm doing a Standalone MOD on my CR-10s, I was running five TMC2130'S on a MKS Gen L gen 1.0 in SPI mode without any issues, when the board was just sitting on desk. After the wife got tired of all those exposed wires, I printed a custom case to neatly make all the component fit inside a fan cool enclosure.

To do that I needed to to resolder the 4 SPI pins on my drivers (SDI, SCK, CK & SDO). The [factory pins](https://i.imgur.com/KlX3rTF.jpg) were pointing straight up and I needed to have them horizontal to fit in the case.

So instead of just bending the pins, I removed the old pins and soldered new [90 deg pins](https://i.imgur.com/RfyyAwr.jpg), thinking is was much cleaner looking. While I had the soldering iron out I added a pin on the "DIAG" for when I'm ready to do sensorless homing.

After fixing a few bad crimps on the Dupont connectors that were giving my "Driver Errors Printer Halted" messages. I was finally ready to home that beauty. But to my horror, none of the axis moved. I resetted the printer many time, but no success. I tried moving just the individual axis manually, no go on that as well...

I've rechecked every solder points they all look good, nothing seems to be shorted out.

Could I have fried the drivers just by re-soldering pins ?

Can someone help me figure out what is wrong ?

[This](https://drive.google.com/open?id=1rvh17SdAd9f9txaqXUX0KzH4f4ofBc_z) is how I've got them wired.

And here is the M122 report:

```
Send: M122
Recv: 		X	Y	Z	Z2	E0
Recv: Enabled		false	false	false	false	false
Recv: Set current	750	750	750	750	750
Recv: RMS current	1325	1325	1325	1325	1325
Recv: MAX current	1868	1868	1868	1868	1868
Recv: Run current	23/31	23/31	23/31	23/31	23/31
Recv: Hold current	4/31	4/31	4/31	4/31	4/31
Recv: CS actual		0/31	0/31	0/31	0/31	0/31
Recv: PWM scale		0	0	0	0	0
Recv: vsense		0=.325	0=.325	0=.325	0=.325	0=.325
Recv: stealthChop	false	false	false	false	false
Recv: msteps		256	256	256	256	256
Recv: tstep		0	0	0	0	0
Recv: pwm
Recv: threshold		0	0	0	0	0
Recv: [mm/s]		-	-	-	-	-
Recv: OT prewarn	false	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false	false
Recv: off time		0	0	0	0	0
Recv: blank time	16	16	16	16	16
Recv: hysteresis
Recv: -end		-3	-3	-3	-3	-3
Recv: -start		1	1	1	1	1
Recv: Stallguard thrs	0	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	Z2	E0
Recv: stallguard
Recv: sg_result		0	0	0	0	0
Recv: fsactive
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0x00:00:00:00
Recv: 	Y = 0x00:00:00:00
Recv: 	Z = 0x00:00:00:00
Recv: 	Z2 = 0x00:00:00:00
Recv: 	E0 = 0x00:00:00:00
Recv: 
Recv: 
Recv: ok
```

teemuatlut, 2018-08-13:
> Could I have fried the drivers just by re-soldering pins ?

Possible but unlikely.
Try getting an example from the library working first with just one motor. If at all possible, with a different board and new simple point to point wiring.

bossman74, 2018-08-13:
Do I start with a clean Marlin and just copy the examples in it ?

bossman74, 2018-08-14:
Ok I tested all five drivers with point to point wiring on my X axis.

4 out of 5 can make the stepper move normally, even connected all the other steppers to that same axis and the all work.  But the last one doesn't do anything at all, no noise or movement...

The four working drivers, all gave the same debug report:
```
Send: M122
Recv: 		X
Recv: Enabled		false
Recv: Set current	825
Recv: RMS current	795
Recv: MAX current	1121
Recv: Run current	25/31
Recv: Hold current	5/31
Recv: CS actual		5/31
Recv: PWM scale	1
Recv: vsense		1=.18
Recv: stealthChop	true
Recv: msteps		16
Recv: tstep		1048575
Recv: pwm
Recv: threshold		0
Recv: [mm/s]		-
Recv: OT prewarn	false
Recv: OT prewarn has
Recv: been triggered	false
Recv: off time		5
Recv: blank time	24
Recv: hysterisis
Recv: -end		2
Recv: -start		3
Recv: Stallguard thrs	0
Recv: DRVSTATUS	X
Recv: stallguard
Recv: sg_result		0
Recv: fsactive
Recv: stst		X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:
Recv: 	X = 0x80:05:00:00
Recv: 
Recv: ok
[...]
```
The fifth one (non working for now) gives me this report:
```
Send: M122
Recv: 		X
Recv: Enabled		false
Recv: Set current	825
Recv: RMS current	1436
Recv: MAX current	2025
Recv: Run current	25/31
Recv: Hold current	5/31
Recv: CS actual		0/31
Recv: PWM scale	0
Recv: vsense		0=.325
Recv: stealthChop	false
Recv: msteps		256
Recv: tstep		0
Recv: pwm
Recv: threshold		0
Recv: [mm/s]		-
Recv: OT prewarn	false
Recv: OT prewarn has
Recv: been triggered	false
Recv: off time		0
Recv: blank time	16
Recv: hysterisis
Recv: -end		-3
Recv: -start		1
Recv: Stallguard thrs	0
Recv: DRVSTATUS	X
Recv: stallguard
Recv: sg_result		0
Recv: fsactive
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:
Recv: 	X = 0x00:00:00:00
Recv: 
Recv: ok
[...]

```

So there's many differences in this report compared to the other ones.

I'm trying to see if there is physically a difference on that driver (bad solder, shorted pins, etc...) but can't find anything.

Do you think this one is fried ???

workheart, 2018-08-14:
So you did use the same socket for the "faulty" as for all the others? If yes it looks like a fried driver. Try to disable stealth chop and do the test with the known working socket and the faulty driver. 
It would still say X = 0x00:00:00:00, so no spi working but the acis would move (lould and without torque) thats the sympthom i got on my burned drivers

bossman74, 2018-08-14:
Yes I was testing them on all the same socket, and the tried connecting each stepper to that socket.  All steppers worked when using the four "good" drivers.

I then realised it was as if the SPI connections were not present on the faulty one.  So even though the solder looked good, I went ahead and redid the solder on my angled pins on the SPI pins of the fifth driver.  And it now that driver works, even in SPI mode !!  I'm glad I did not damage it.

But I'm not out of the woods yet.  I tried reinstalling all of them in each of their own socket connected with my previously working wire harness, and was back with my initial issue.  No driver error, but no movement either.  So to confirm that my harness was bad, I went back to a basic Marlin configuration defining each axis one at a time.  And made sure that each driver/stepper combination work one at a time with simple point to point connections.  They all do, so now I'm making a new wire harness to hook all five drivers to the SDI, SCK and SDO pins of my board.

At least I'm making progress ;-)

By the way, is it normal for steppers to make noise in spread cycle mode, even when then are not moving ???  They were actually more noisy than my old stock drivers on the CR-10s...  Anyway I've switched back to the blissful quietness of stealthchop :-)

teemuatlut, 2018-08-14:
> is it normal for steppers to make noise in spread cycle mode, even when then are not moving

Lets say the current configuration is not quite ideal and on 12V the chopper frequency is clearly in the audible range. But I've implemented ways to tune the drivers with the help of an example sketch that comes with the library and a new configuration parameter. This however is not yet supported in upstream Marlin.

bossman74, 2018-08-15:
Success !

Made a new serial wire harness and I'm back in business.  So I had two issues, a bad solder on one driver and a bad connection from one of the wires in my previous harness.  Both problems were caused by my lack of talent in small electronics... :-(

Now everything is neatly packaged in my enclosure and the wife is "satisfied" with the final result.  Happy wife, happy life!

My next goal is to get sensorless homing working to get rid of my homing limit switch.

Thank you teemuatlut and workheart for your help, you can close this issue.

closed 2018-08-15 by teemuatlut