teemuatlut/TMC2130Stepper #53: fCLK - where to get this value

jiri-jirus, 2018-09-08:
Hello, i have a question - how can i get information about fCLK of particular SilentStepStick? this value (fCLK) is used in many formulas, but i dont know whether it is set by some register or hardwired by oscillator on stepstick...? thank you very much for explanation.

teemuatlut, 2018-09-08:
Datasheet, p88

# 26 Clock Oscillator and Input

The clock is the timing reference for all functions: the chopper and the velocity thresholds. Many parameters are scaled with the clock frequency, thus a precise reference allows a more deterministic result. The on-chip clock oscillator provides timing in case no external clock is easily available.

## USING THE INTERNAL CLOCK

Directly tie the CLK input to GND near to the IC if the internal clock oscillator is to be used. For best precision, the internal clock can be calibrated by reading out TSTEP at a defined external step frequency. It is easiest to use 1 kHz microstep frequency. With this, TSTEP shows the number of internal clock cycles per millisecond, i.e. TSTEP=1200 means that fCLK is 12 MHz in the actual IC. Scale velocity thresholds, TOFF and PWM_FREQ based on the determined frequency. Temperature dependency and ageing of the internal clock is comparatively low.

---
I can't remember if that's the corrected information or not, but using this I've determined the f_clk to be around 12.5MHz.
The instructions there might be off by an order of 10.

jiri-jirus, 2018-09-08:
thank you. so this M122 output means that i have 10.5MHz fCLK? 

`Send: M122
Recv: 		X	Y
Recv: Enabled		true	true
Recv: Set current	400	600
Recv: RMS current	397	581
Recv: MAX current	560	819
Recv: Run current	12/31	18/31
Recv: Hold current	6/31	9/31
Recv: CS actual		6/31	9/31
Recv: PWM scale	34	42
Recv: vsense		1=.18	1=.18
Recv: stealthChop	true	true
Recv: msteps		16	16
Recv: tstep		1048575	1048575
Recv: pwm
Recv: threshold		79	121
Recv: [mm/s]		100.08	65.34
Recv: OT prewarn	false	false
Recv: OT prewarn has
Recv: been triggered	false	false
Recv: off time		5	5
Recv: blank time	24	24
Recv: hysterisis
Recv: -end		2	2
Recv: -start		3	3
Recv: Stallguard thrs	0	0
Recv: DRVSTATUS	X	Y
Recv: stallguard
Recv: sg_result		0	0
Recv: fsactive
Recv: stst		X	X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:
Recv: 	X = 0x80:06:00:00
Recv: 	Y = 0x80:09:00:00
Recv: 
Recv: ok`

teemuatlut, 2018-09-08:
No.
You can't do it in Marlin.

jiri-jirus, 2018-09-08:
i went off: " With this, TSTEP shows the number of internal clock cycles per millisecond, i.e. TSTEP=1200 means that fCLK is 12 MHz in the actual IC."
and this part of M122: "Recv: tstep 1048575 1048575" 
I know I cant change fCLK in Marlin, but I can get info about actual fCLK, right?

teemuatlut, 2018-09-08:
# TSTEP

Actual measured time between two 1/256 microsteps derived from the step input frequency in units of 1/fCLK. Measured value is (2^20)-1 in case of overflow or stand still.
The TSTEP related threshold uses a hysteresis of 1/16 of the compare value to compensate for jitter in the clock or the step frequency: (Txxx*15/16)-1 is the lower compare value for each TSTEP based comparison.
This means, that the lower switching velocity equals the calculated setting, but the upper switching velocity is higher as defined by the hysteresis setting.

---

2^20 - 1 = 1048575 == standstill

jiri-jirus, 2018-09-08:
well, my tstep now shows 759 in the middle of move with X axis ( G1 X200 F600 I have M92 X100 = 1kHz microstep frequency) that seems to be 7.5MHz fCLK?

teemuatlut, 2018-09-08:
> Actual measured time between two 1/256 microsteps derived from the step input frequency

`TSTEP` is _not_ the time between your input `STEP` signals. Unless you're running with 256 ms setting.

I'd suggest you make an external Arduino sketch to test your drivers.
According to the datasheet instructions you should end up with a figure something like 1.2MHz. Multiply that by 10 to get your actual f, because the information given is not quite correct.
**IIRC**

By your calculations, you say you'd have 7.5MHz f_clk.
If we divide this by 10, because of the misinformation, we get 750kHz.
This is the time between two **interpolated** (think in terms of 256 microstepping even if that is not the setting) step pulses. The ratio between 16 (assumption) and 256 is 16.
Therefore 750kHz * 256/16 = 12MHz