teemuatlut/TMC2130Stepper #54: Got TMC2130Stepper library working with STM32 on Arduino IDE

ASMotionLab, 2018-10-08:
Great library, thank you.

In Arduino IDE, I tried to compile my code with TMC2130Stepper for my 'Blue Pill' STM32F103 (hardware support from stm32duino.com).

Ran into a load of compiler errors, pointing to the use of 'constexpr' in TMC2130Stepper_REGDEFS.h. I don't have the error log now, but it was something to do with compiler version not understading constexpr.

I replaced every instance of 'constexpr' with 'const' in TMC2130Stepper_REGDEFS.h, and it compiled fine and works fine.

I couldn't replicate this by installing the whole toolkit from fresh along with all the hardware definitions, boards, libraries on a new machine. This error doesn't show up. But it did on my older dev computer that's had incremental arduino updates for the last 5 years.

So just in case anyone's looking for this issue - it might help someone to get their prototype working.

Not sure about the actual implications of using const vs constexpr, maybe someone could explain?

teemuatlut, 2018-10-09:
In a nutshell, `constexpr` means the value can be calculated and used in compile time. Sort of like type safe `#define`. Const just tells the compiler that the value will/should not be changed at any point, and you'll get a compile error if you try to do so. `Const` is more just for the developer's aid but a smart compiler _can_ do some optimizations based on the keyword.

Without an error log I can't say much about using the library with STM32.

knutollebolle, 2019-02-23:
I think I've got the same errors:

/Users/svenklein/Documents/Arduino/libraries/TMC2130Stepper/src/TMC2130Stepper_REGDEFS.h:33:11: error: 'uint8_t' does not name a type
 constexpr uint8_t REG_DCCTRL  = 0x6E;

It will give this error on almost every line from the TMC2130Stepper_REGDEFS.h file. I'm working with Arduino 1.8.8 on a MacBook. I hope you can help me out, at the moment I'm not able to get my TMC2130 working on a MKS Gen L V1.0

knutollebolle, 2019-02-24:
Problem solved, I had to take out the TMC2130Stepper.h from the header. No problems with compiling now.