teemuatlut/TMC2130Stepper #56: TMC2130: SDI and SDO swapped in Fritzing PCB view

eyer1951, 2018-11-05:
In the PCB definition, the Fritzing part for TMC2130 has SDO and SDI swapped.  I had some boards manufactured, unfortunately.

In this example i just connected DIR to SDO to illustrate the issue.
[bad2130pcb.zip](https://github.com/teemuatlut/TMC2130Stepper/files/2550027/bad2130pcb.zip)

teemuatlut, 2018-11-05:
@soswow

eyer1951, 2018-11-05:
Oy!  CS and SCK are also swapped.

teemuatlut, 2018-11-05:
I'm not that familiar with Fritzing but everything looks good to me. Something I'm missing?

eyer1951, 2018-11-05:
Thanks for checking. Can you open the sketch file I provided?  If so, you will quickly see that the PCB design does not match the TMS2130 icon shown in the breadboard (part layout). Pin locations for PCB wiring don't match the part. The location of the SDO connection (should be in the middle of the row) in the PCB layout is shown (wrongly) as 2nd from the end of the part (location of SDI).  There is a simplified parts editor in Fritzing, but i could not fix this. Also note, CS and SCK are swapped. It appears @soswow was the author of the Fritzing parts file.

teemuatlut, 2018-11-05:
Now I see the issue!
I think the order of named pins need to be changed in the Connecters tab.

eyer1951, 2018-11-05:
I tried renaming them in the Connectors tab, but that didn't work. I couldn't get Schematic, Breadboard, and PCB to synchronize correctly.
'

eyer1951, 2018-11-06:
I did figure out how to associate each connections with the "icon" (holes on the part). I had not seen how to enable the menu with the checkmarks and "select graphic." So, for my own use, i now have a part that works. It would be nice if the "official" one were to be corrected and re-published.

soswow, 2018-11-06:
I'll look into that today. Sorry for swapping (

soswow, 2018-11-06:
raised a PR