teemuatlut/TMC2130Stepper #58: Looking for Help With StallGuard

Legohead259, 2018-11-08:
Good day. I've gotten the StallGuard example to work on my Arduino Mega, however, I am confused on how to properly read if the STALL flag has been triggered. I've been able to Jerry-Rig it with `(drv_status & SG_RESULT_bm)>>SG_RESULT_bp <= 5` to disable the motor but I know there has to be a cleaner way to get this value. Is there a function I can call to get the STALL flag without that clunkiness?

Also, is there any better documentation on the timer interupt system that is running the motor in the StallGuard example? I'd like to implement it, but I have no idea how it works past `TIMSK1` is the clock and `OCR1A` is the clock speed.

 Thanks!

dmitrif, 2018-11-08:
You can write a function that wraps the bitshifting to read the flag from
the registry but it’s pretty standard in embedded programming so just
abstract it out if needed.

For the timers / interrupts read an AVR tutorial in regards to timers and
interrupts. This is not specific to this library and is common for most
microprocessors implementing interrupts.

teemuatlut, 2018-11-08:
To get the stallguard flag
```cpp
bool sg_status = driver.stallguard();
```
To get the stallguard measurement
```cpp
uint16_t sg_measurement = driver.sg_result();
```

dmitrif, 2018-11-08:
> To get the stallguard flag
>
> bool sg_status = driver.stallguard();
>
> To get the stallguard measurement
>
> uint16_t sg_measurement = driver.sg_result();

Learned something new, thanks @teemuatlut!
_________________

Dmitri Farkov
647.898.5054

pilhuhn, 2019-01-03:
Should the motor be turning in that example, as I see no `digitalWrite(STEP_PIN,..);` commands

teemuatlut, 2019-01-03:
Yes. AVR timer interrupt:
```cpp
ISR(TIMER1_COMPA_vect){
  PORTF |= 1 << 0;
  PORTF &= ~(1 << 0);
}
```