teemuatlut/TMC2130Stepper #60: Problem with Running for Certain Number of Steps

Legohead259, 2018-11-13:
Hello,
I recently modified the Simple.ino example so that I could get my motor to step enough times to make one full revolution (200 in my case). However, I am having issues where I have to increment my step counter by some weird factor (1/250) just to get close to one revolution. I don't know what is causing this and the issue is not prevalent on my simpler stepper controllers. Any help would be appreciated.

teemuatlut, 2018-11-13:
What's your microstep setting?

Legohead259, 2018-11-13:
I have it set to 1 right now. Experimented with 16 and 32 but they also
require weird increments to work properly. Is there an equation I'm missing
to get the number of steps correct?

teemuatlut, 2018-11-13:
If you have a 200 fullsteps motor and your microstep setting is 16, then you need to send `200*16=3200` step pulses for a full rotation.

Legohead259, 2018-11-13:
Ok. I just tried it. For some reason at 16 microsteps, the motor just goes
crazy; it starts whining and stuttering. It works just fine at 32
microsteps though. Any idea what could be causing that?

teemuatlut, 2018-11-13:
Possibly bad config being pushed for some reason.
Any time there's a read command, the library will remember the response and base the subsequent writes on that data pattern. If the read was faulty for some reason, it can affect the following writes.

Legohead259, 2018-11-13:
Interesting. Is there much of a torque difference between the microsteps
(i.e. more steps=more torque) that yo know of?

teemuatlut, 2018-11-13:
The relation with microstepping and motor torque is a myth/misunderstanding.
All that matters is the angle between the rotor and the electric field (and the winding current).
You often hear that increasing microstep settings decreases the motor torque, but no one ever remembers that you also take twice as many steps for the same distance/rotation and thus the electric field ends up the same.

Increasing the number of microsteps will not get you more torque, but will result in smoother motion. This is what the Trinamic microPlyer (=microstep interpolation) does.

Legohead259, 2018-11-14:
I see. Thank you for your assistance. Another follow up question: what's
the best way to adjust the motor speed? Is there a dedicated register or
method I can use?

teemuatlut, 2018-11-14:
Control the time between sent step pulses.

Legohead259, 2018-11-14:
Excellent. Thank you