teemuatlut/TMC2130Stepper #62: Reading DRV_STATUS

gvandersel, 2018-12-18:
I have the Simple example working (version TMC2130stepper 2.5.0) and the steppermotor is stepping in two directions under HW-SPI control. I'm using a Arduino pro Mini (5V, 16 Mhz) for controlling the TMC2130. It is a chines clone (Fysetc) with SPI enabled and CFG4 tied to GND and CFG5 tied to VIO (+5V).
When i run Simple with the additional lines in the if clause
    Serial.println(TMC2130.version(), HEX);
    Serial.println(TMC2130.DRV_STATUS(), HEX);
I get the following readings:
(No stepper attached)
Dir -> 0
11
60120000  or
Dir -> 1
11
60120000 
(Stepper attached)
Dir -> 0
11
12000000  or
Dir -> 1
11
12000000

I have red the TMC2130 datasheet (chapter 5.5.5, page 38), but I can't figure out what I'm reading.

Does anyone have a clue?

Groet,

Gerard van der Sel (NL)

teemuatlut, 2018-12-18:
p26: register IOIN:
```
VERSION: 0x11=first version of the IC
Identical numbers mean full digital compatibility.
```

For the DRV_STATUS:
`0x60120000` == `0b0110 0000 0001 0010 0000 0000 0000 0000`
Meaning bits 17, 20, 29 and 30 are true.
These correspond to `CS_ACTUAL = 0b10010 = 18` and
open load indicator phases A and B are true. Which makes sense if you have no stepper attached. These two also went away when you attached a motor.

gvandersel, 2018-12-18:
Thanks,

If I understand it correct the reading with the stepping stepper is:
0x12000000 == 0b0001 0010 0000 0000 0000 0000 0000 0000, which means:
s2gb set (phase B shorted to ground)
otpw set (pre warning overtemperature)
If I understand correctly the driver should be disabled (stepper not stepping?) because of s2gb.

Groet,

Gerard van der Sel (NL)

closed 2019-01-03 by gvandersel