teemuatlut/TMC2130Stepper #65: Setup in memory and then push

gvandersel, 2019-01-03:
Is it possible to do the setup for the TMC2130 in memory and use a push to send all information to the TMC2130. Wenn I looked through the library all register settings end in a send2130. It is a waste of time in this way.

Groet,

Gerard van der Sel (NL)

teemuatlut, 2019-01-03:
You can form the full 32bit register pattern and then push it as a whole.

```cpp
uint32_t my_gconf = 0;
my_gconf |= 1 << EN_PWM_MODE_bp;
my_gconf |= 1 << SHAFT_bp;
...
driver.GCONF(my_gconf);
```

gvandersel, 2019-01-04:
It could work, but i have to use a shadow of the shadow in the driver. But the code should be changed to:

// Get current value
uint32_t gconf = driver.GCONF();
// Apply changes
gconf |= (1 << EN_PWM_MODE_bp);   // Set bit
gconf &= ~(1 << SHAFT_bp);              // Clear bit
...
// Send new value to TMC2130 and update driver value
driver.GCONF(gconf);

It has the advantage of sending only the changed values and not all registers with driver.push();
I hadn't found the bitdefinitions in the source, but now i see you have defined the masks also.