teemuatlut/TMC2130Stepper #68: How to control with SPI Mode? (without STEP Pin) Example needed

martinmatias, 2019-02-04:
Hi All, I want to move stepper motor through control TMC2130. I undestand that it can be done using using only SPI (I mean without EN, STEP and DIR PINS). How that possible?

After setting direct_mode to 1 dont know how to move the stepper.

I have succesfully run all examples, I have SPI connection OK (using arduino UNO and ESP8266) and checked SPI connection (it works OK)

Regards
Martin

teemuatlut, 2019-02-04:
With direct mode you have to control the coil currents manually using the `XDIRECT` register, or with methods `coil_A` and `coil_B`. What values to use becomes clear when you examine how the stepper motor works in principle.

By the sound of it, you'd likely rather want to use a driver from the TMC5xxx range and make use of the ramp generator, but I don't know if this is a possibility for you.

martinmatias, 2019-02-04:
Hi, thanks for your reply. I'm working on a project (blinds control, nothing to do with 3D printer). Until I found TMC2XXX I was using pololu a4988 controlled using ESP8266 wifi module. a4988 is a little bit noisy and needed sensors at both ends (I was using hall sensor effect, whitch require a cable trough all the blind length).
I thought that TMC2130 would reduce noise and with Stallguard2 can avoid sensors.
So, can you provide few examples about how to use XDIRECT register? If I have to be switching betwen coil_A and coil_B per every movement I think it would not be possible.

If not, is there any example using stallguard2 and a library like AccelStepper.h?
TMC2130_AccelStepper example dont allow to use stallguard (I think).
Also, I've succesfully made stallguard example work on Arduino UNO, but then I've tried to modify code to move through loop (like simple example) but see that DRV_STATUS() value didnt change anymore (like many issues posted here regarding Interrupts).

Any idea or solution would be appreciated.
Regards
Martin

martinmatias, 2019-02-04:
OK, I agree with you. If stallguard only can be used with STEP signal only, SPI mode is not what I'm looking for,
I'll try placing stepper.run() inside ISR and then comment into another user for other if they want to use it. I'll close this issue.
Regards
Martin

closed 2019-02-04 by martinmatias

elavarasibaskar, 2021-01-22:
Hi, 

   Do you have any update on using the stepper.run() inside ISR to enable StallGuard cause I also have the same kind of application where I need to use Coolstep, StallGuard feature and also enabling SPI mode from where I can control both Step and Direction without using Step and Direction pin externally.

teemuatlut, 2021-01-22:
It sounds more like you should rather be using the TMC5160 / 5161 / 5130, which support the internal ramp generator.

The TMC2130 isn't really well suited to be used without step and dir lines.

elavarasibaskar, 2021-01-22:
@teemuatlut Thanks for the info.

julian-weinert, 2021-01-26:
I'm really sorry to look for help in a GH issue, but I'm kind of hopeless. As far as I understand section 1 (page 5) of the datasheet correctly, I should be able to use the step/dir interface and do configurations via SPI.
I have a SilentStepStick (to familiarise with the chip) which is shipped with `SPI_MODE` tied to ground. Meaning it operates in stand alone mode.

Now I removed the jumper to enable SPI config and since then the step interface seems to be disabled.
Did I simply miss something? Do I need further setup in this mode to get up and running?

Thanks and sorry again for posting here...