teemuatlut/TMC2130Stepper #70: Enable/Disable driver via SPI

orkunkilinc, 2019-03-13:
Hi guys,

I did not route the EN pin of the stepper on my PCB thinking I won't need it and did not have enough pins either.

Now there is a scenario where I need to disable the driver so the motor can spin freely by hand.

I read the datasheet for tmc2130 and also went through the registers in the library but couldn't come up with something.

I tried setting the both rms_current and hold_current multiplier to 0 but the coils are still energized. I know the driver does not turn on before we call begin() so there should be a way to disable it somehow.

Any ideas?
Regards
Orkun

teemuatlut, 2019-03-13:
Set off time to 0 to disable the driver.

orkunkilinc, 2019-03-13:
Awesome, thank you!

I'm closing the issue, but while fiddling with off_time I realized on the main page function descriptions it says:

> Sets off_time = 2 and blank_time = 24

but in TMC2130Stepper.cpp under begin() it says:

> toff(8); //off_time(8);
	tbl(1); //blank_time(24);

Just a heads up.

closed 2019-03-13 by orkunkilinc