teemuatlut/TMC2130Stepper #73: TMC 5072 Stallguard problem

SrinivasD18, 2019-03-29:
Currently I am working on “TMC5072-BOB” StallGuard2™ feature connected to  STM32 board. 
Could you please assist me how to implement “sensorless homing” position. 

void TMC5072 Init()
{
	if(tmc5072_readInt(&TMC5072, TMC5072_GSTAT)!= 0x01 )
	{
		return ; // error here 
			
	}

    tmc5072_writeInt(&TMC5072, TMC5072_GCONF, 0x00000008 );

	for(u8 motor = 0; motor < TMC5072_MOTORS; motor++)
	{
	tmc5072_writeInt(&TMC5072, TMC5072_CHOPCONF(motor), 0x000101D5);
	tmc5072_writeInt(&TMC5072, TMC5072_IHOLD_IRUN(motor), 0x00071400 );
	tmc5072_writeInt(&TMC5072, TMC5072_VHIGH(motor), 0x00FFFFFF       );
	tmc5072_writeInt(&TMC5072, TMC5072_PWMCONF(motor), 0x0000EA60);
	tmc5072_writeInt(&TMC5072, TMC5072_VCOOLTHRS(motor), 30000 );
	tmc5072_writeInt(&TMC5072, TMC5072_TZEROWAIT(motor), 0x0000000A);

	tmc5072_writeInt(&TMC5072, TMC5072_RAMPMODE(motor), TMC5072_MODE_POSITION);
	tmc5072_writeInt(&TMC5072, TMC5072_XTARGET(motor), 0x00);
	tmc5072_writeInt(&TMC5072, TMC5072_XACTUAL(motor), 0x00);

	//SaveTMCDefaultSettings_Motor(motor);
	//SAPMotorConfig(motor);

	tmc5072_writeInt(&TMC5072, TMC5072_VSTART(motor), 1);
	tmc5072_writeInt(&TMC5072, TMC5072_A1(motor), 5000);
	tmc5072_writeInt(&TMC5072, TMC5072_V1(motor), 26843);
	tmc5072_writeInt(&TMC5072, TMC5072_AMAX(motor), 5000);
	tmc5072_writeInt(&TMC5072, TMC5072_VMAX(motor), 200000);
	tmc5072_writeInt(&TMC5072, TMC5072_DMAX(motor),5000);
	tmc5072_writeInt(&TMC5072, TMC5072_D1(motor),5000);
	tmc5072_writeInt(&TMC5072, TMC5072_VSTOP(motor),10);

	vmax_position[motor] = motorToIC(motor)->config->shadowRegister[TMC5072_VMAX(motor)];

	}
}
--------------
All the motor movement works fine without any noise.
For the StallGuard2™ feature:

vmax = 2000		//max. speed (0 ... 2047)
amax = 1000	                 //max. acceleration (0 ... 2047)
cmax = 23		//max. current 
stallGuardValue = 48 		//Calculated from Stallgrauard tool
coolStepThresholdSpeed = 1000	//min. speed for coolStep™ (0 ... 2047)
coolStepHysteresisWidth = 5     	//coolStep™ hysteresis width (0 ... 15)
coolStepHysteresisStart = 1	         //coolStep™ hysteresis start (0 ... 15)
coolStepUpStep	= 3	     //coolStep™ up step setting (0 .. 3): 0 = tiny ... 3 = large
coolStepDownStep = 3	//coolStep™ down step setting (0 ... 3): 0 = slow ... 3 = very fast
 
//Set actual position as the zero position
SAP (0, 0, 0);
SAP (1, 0, 0);

//Switch off stallGuard
SAP (181, 0, 0);

// *** Interrupt initialization *** //
SAP (5, 0, amax);						//set max. acceleration
SAP (6, 0, cmax);    					//set max. current
 
// *** stallGuard™ initialization *** //
SAP (173, 0, 0    	);		//set stallGuard™ filter disable: 0 = disable, 1 = enable
SAP (174, 0, stallGuardValue  );	//stallGuard™ threshold
SAP (181, 0, 1 	);		       //min. speed for motor to be stopped by stallGuard™
 
// *** coolStep™ initialization *** //
SAP (168, 0, 1    );		//set coolStep™ minimum current setting: 0 = 1/2, 1 = 1/4
SAP (169, 0, coolStepDownStep  );		//set coolStep™ down step setting
SAP( 170, 0, coolStepHysteresisWidth	);//set coolStep™ hysteresis width
SAP (171, 0, coolStepUpStep    );		//set coolStep™ up step setting
SAP (172, 0, coolStepHysteresisStart	//set coolStep™ hysteresis start
SAP( 182, 0, coolStepThresholdSpeed	);//set coolStep™ threshold speed
SAP (183, 0, 23   ); 					//set coolStep™ slow run current

ROR 0, 1000 //Let the motor run until obstructed
While (1)
{
    If(  GAP 206, 0 ) == 0 ) //See if the motor has been stopped due to a stall
    stop(0); // Stop Motor 1 
}

I tried with SAP (181, 0, 1 	); and SAP (181, 0, 0 ); == > both cases not working .

It’s not working. Could you please let me know how to proceed with this.

teemuatlut, 2019-04-21:
I'd suggest you look into how the TRAMS platform handled homing. My branch is getting really old at this point but I seem to remember that I did have the homing procedure working.
https://github.com/teemuatlut/Marlin/blob/1f5652a0c22eff564e81e24f1719deae67760367/Marlin/TRAMS.cpp#L77-L249

Good luck with your project!

SrinivasD18, 2019-08-13:
Seems like its no working with TMC5072 ChipSet