teemuatlut/TMC2130Stepper #74: TMC write is OK, but read doesn't work. the data always be 1111111111111111 or 0xFFFFFF

ziyangwang007, 2019-04-22:
Hi!

Thanks for creating this excellent and detailed library for TMC2130. 

It saves lots of time for us.

I only have one problem when I use this library, my code or test other people's code.

For me, write and sending data works well, I can easily test and control my motor, but when I try to read the data and status from the register by SPI.

The results always be 11111111111111111111111111111111 or 0xFFFFFFFF(it is same). and 8 bit SPI Status always be 0. 

Do you have any ideas?

does it means some registers should be configured? or some other problems need be fixed?

Thanks!

teemuatlut, 2019-04-22:
It's quite possible for the MISO line to have issues while MOSI works just fine.
It's a bit broad advice but check the wiring, grimps and soldering. Also make sure the driver has VMOT power from the start.

ziyangwang007, 2019-04-22:
I appreciate your quick and professional reply!

Yes, I connect the SilentStepStick and Arduino Mega board with new wires and breadboard. It works now. I can read and write data to the registers.

In further, I realized when I use "Serial.print" function to read status, the motor may stop working for print. After the end of print, the motor start to work again.

And also, the status and data I read from motor never change. 
For example, DRVSTATUS always be  111000101000000001010000000000, even when I put mechanical load.

Do you have any idea for that? thank you again for your time!

teemuatlut, 2019-04-22:
Move the step pin write calls to an interrupt routine so that the stepping takes priority over the rest of the program.
When normally running a motor at a constant speed (driver doesn't go into hold current), the only value that is likely to change is `SG_RESULT`. However, this only applies when using spreadCycle and also needs tuning to get a sensible value.

closed 2019-04-26 by ziyangwang007