teemuatlut/TMC2130Stepper #77: typo's in TMC2130Stepper.h / TMC2130Stepper.cpp (section XDIRECT)

djamu, 2019-04-26:
Hi, first of all thanks for this library.

I found a couple of typo's in TMC2130Stepper.h / TMC2130Stepper.cpp while trying XDIRECT

TMC2130Stepper.h  line 296 (// RW: XDIRECT)
should be :
`		// RW: XDIRECT
		inline int16_t 	coil_A_current()									__attribute__((always_inline)) { return coil_A(); 									}
		inline void 		coil_A_current(		 int16_t value)	__attribute__((always_inline)) {				coil_A(value); 							}
		inline int16_t 	coil_B_current()									__attribute__((always_inline)) { return coil_A(); 									}
		inline void 		coil_B_current(		 int16_t value)	__attribute__((always_inline)) {				coil_B(value); 							}
`
TMC2130Stepper.cpp line 239 should be 
`// RW: XDIRECT
uint32_t TMC2130Stepper::XDIRECT() { TMC_READ_REG(XDIRECT); }
void TMC2130Stepper::XDIRECT(uint32_t input) {
	XDIRECT_sr = input;
	TMC_WRITE_REG(XDIRECT);
}
void TMC2130Stepper::coil_A(int16_t B) 	{ TMC_MOD_REG(XDIRECT, COIL_A); 	}
void TMC2130Stepper::coil_B(int16_t B) 	{ TMC_MOD_REG(XDIRECT, COIL_B); 	}
int16_t TMC2130Stepper::coil_A() 		{ TMC_GET_BYTE_R(XDIRECT, COIL_A); 	}
int16_t TMC2130Stepper::coil_B() 		{ TMC_GET_BYTE_R(XDIRECT, COIL_B); 	}
`

I also made the library ESP32 compatible (for either SPI ports), let me know if you're interested.

teemuatlut, 2019-04-26:
I reformatted your comment for better readability.

If you want, you can put together a pull request and show what changes are needed.
But I also should point out that this library has been superseded by the `TMCStepper` library.

djamu, 2019-04-26:
Thanks for the edit, forgot how reformat properly.
just saw there's yet another typo. 
```
inline int16_t coil_B_current() __attribute__((always_inline)) { return coil_A(); }

```
should obviously be 
```
inline int16_t coil_B_current() __attribute__((always_inline)) { return coil_B(); }
```

teemuatlut, 2019-05-01:
https://github.com/teemuatlut/TMC2130Stepper/releases/tag/v2.5.1

closed 2019-05-01 by teemuatlut