teemuatlut/TMC2130Stepper #78: set current

ziyangwang007, 2019-05-11:
Thanks for your work on TMC2130 library.

I feel confused about setting current.

I find two methods to set current. 

One method is to write the "irun“ and ”vsense “ register. and then the "Irms" could be calculated by:
 (float)(irun()+1)/32.0 * (vsense()?0.180:0.325)/(Rsense+0.02) / 1.41421 * 1000;

The other method is for SilentStepStick2130 by measuring the voltage on the Vref pin and adjusting the voltage with the potentiometer.
I achieve it follow by the instruction in other website: https://learn.watterott.com/silentstepstick/faq/
https://wiki.fysetc.com/TMC2130/

The question is that what is the connection between these two methods, and the meaning of  the mathematical expression in your library.

Thank you!

teemuatlut, 2019-05-11:
`irun` and `ihold` set the registers directly and you'll need to calculate the CS values yourself, also accounting for selecting the correct vsense setting.

`SilentStepStick2130` is a shorthand for setting the current with default hold multiplier and default Rsense value from Watterott TMC2130 design. This can also be achieved by just calling `rms_current(mA)`. It does not measure the voltage on Vref.
https://github.com/teemuatlut/TMC2130Stepper/blob/188d5fe35d4406a8e342737dc0107046d1ce1585/src/TMC2130Stepper.h#L17

You can see the logic behind the math in here:
https://github.com/teemuatlut/TMC2130Stepper/blob/188d5fe35d4406a8e342737dc0107046d1ce1585/src/source/TMC2130Stepper.cpp#L343-L371

The easiest way to set the current is just with `rms_current(mA)` or `setCurrent(mA)`.

ziyangwang007, 2019-05-11:
Thanks for your reply!

Yes, for software part, I could set current by these functions: setCurrent() or SilentStepStick2130() or run_current() or irun().

But after I completed the software part to set current, I could still change the current by changing hardware part. I could change the resistance of a resistor on the SilentStepStick2130 board (Vref could be changed).

So it makes me confused.

> `irun` and `ihold` set the registers directly and you'll need to calculate the CS values yourself, also accounting for selecting the correct vsense setting.
>
> `SilentStepStick2130` is a shorthand for [...]

teemuatlut, 2019-05-11:
If you've enabled `I_scale_analog`, then the current is also scaled according to Vref.

closed 2019-05-13 by ziyangwang007