teemuatlut/TMC2130Stepper #80: Software SPI not working

TillScout, 2019-06-13:
I use the library with an Arduino Uno. `Simple.ino` works exactly as expected. I use this pin setup:
```
#define EN_PIN    7  // Nano v3:	16 Mega:	38	//enable (CFG6)
#define DIR_PIN   6  //			19			55	//direction
#define STEP_PIN  5  //			18			54	//step
#define CS_PIN    8  //			17			64	//chip select

```
But when using `Software_SPI.ino`, I get no reaction. Since the pin setup is very similar to the above:
```
#define EN_PIN    7  // Nano v3:	16 Mega:	38	//enable (CFG6)
#define DIR_PIN   6  //			19			55	//direction
#define STEP_PIN  5  //			18			54	//step
#define CS_PIN    8  //			17			64	//chip select
#define MOSI_PIN  9
#define MISO_PIN 10
#define SCK_PIN  11
```
I can rule out wrong connections (just have to move MOSI, MISO and SCK two pins to the right to the hardware SPI pins and load `Simple.ino`).

What happens, when using software SPI is that the stepper turns by very little (possibly one step) and then nothing else happens. On the serial monitor I get this output:
`DRV_STATUS=0b10000000000010010000000000000000`
I did not modify any of the files, apart from pin definitions.

Tamonir, 2019-08-27:
Hi,

same Problem here. I use a Fysetc 2130 driver, unsoldered/resoldered vor SPI mode in my 3D Printer - at least there it is working. Connecting the driver to the Arduino nano as described In SPI example, I don't get anything at all. Is it correct, that steps are still done by step pin and not by SPI?

When I resolder the CFG bridge to standalone mode, the stepper is at least making some noise but still no movement. 

Do you have any idea? How have the solder points on the tmc have to be soldered?

So far, thanks in advance, 

Christian

changedsoul, 2020-07-12:
Almost a year later and still this doesn't work and there is no answer?
At least a nice response of "The code doesn't work, were not planning on fixing it any time soon, figure it out for yourself" would be better than nothing. Us who are beginning this don't know if its the code that's wrong, or what were doing is wrong, or how we are implementing it, etc.

A year later and I am facing the same problems. I am lost how to proceed. I was hoping for a working example to learn from.