teemuatlut/TMC2130Stepper #81: Trouble getting simple sketch example working ver 2.5.1

Mogwai77, 2019-07-21:
Wondering if I could get some help.
I am just mucking around getting familiar with the TMC2130 library in direct mode (not SPI) at this stage.
Thought to just start with the simple example to make sure things were working ok.
Turns out I cannot get the direction change to work using original code as disclosed within the simple example.
I can get the direction to change by modifying the code to use the Boolean logic originally in the code example, I just use the 'dir' to provide the logic state for the digitalWrite on the DIR_PIN.

The code I am using and works is shown below, I have commented out the original driver.shaft_dir(0) command (but its still there)
Hardware is a Arduino Nano (original bootloader), FYSTECH V1.0 TMC2130 driver and Keyestudio V4 CNC board (only used to provide a convenient work platform).

To note: although I have assigned the CS pin to digital pin 12, it's not actually connected.
Kind regards,
Jamie

/**
 * Author Teemu Mäntykallio
 * Initializes the library and turns the motor in alternating directions.
*/

#define EN_PIN    8   //enable (CFG6)
#define DIR_PIN   5  //direction
#define STEP_PIN  2 //step
#define CS_PIN    12 //chip select

bool dir = true;

#include <TMC2130Stepper.h>

TMC2130Stepper driver = TMC2130Stepper(EN_PIN, DIR_PIN, STEP_PIN, CS_PIN);

void setup() {
	Serial.begin(9600);
	while(!Serial);
	Serial.println("Start...");
	driver.begin(); 			// Initiate pins and registeries
	driver.rms_current(600); 	// Set stepper current to 600mA. The command is the same as command TMC2130.setCurrent(600, 0.11, 0.5);
	driver.stealthChop(1); 	// Enable extremely quiet stepping
	
	digitalWrite(EN_PIN, LOW);
        digitalWrite(DIR_PIN, LOW);
}

void loop() {
	digitalWrite(STEP_PIN, HIGH);
	delayMicroseconds(45);
	digitalWrite(STEP_PIN, LOW);
	delayMicroseconds(45);
	uint32_t ms = millis();
  static uint32_t last_time = 0;
  if ((ms - last_time) > 10000) {
		if (dir) {
			Serial.println("Dir -> 0");
			digitalWrite(DIR_PIN, dir);
			//driver.shaft_dir(0);
      
		} else {
			Serial.println("Dir -> 1");
			digitalWrite(DIR_PIN, dir);
			//driver.shaft_dir(1);
 
		}
		dir = !dir;
		last_time = ms;
	}
}

teemuatlut, 2019-07-21:
With "direct mode" do you mean stand alone mode? Because direct mode is an actual feature within the driver.

In stand alone mode the driver works with default values and just the step/dir/en interface with no SPI lines.
`shaft` is an SPI command that obviously therefore doesn't work in standAlone mode. Neither does any of the other SPI commands.

Basically if you're using the drivers in stand alone mode only, then you don't even need to use this library =)

Mogwai77, 2019-07-21:
Ah, makes sense, yes Stand alone mode :-)
Using Simple code as the basis, below was the min I needed to get it to work.
CS still defined but not used.
Thank-you for your help teemuatlut

Kind regards,
Jamie

/**
 * Author Teemu Mäntykallio
 * Initializes the library and turns the motor in alternating directions.
*/
// pins to use Keyestudio V4 CNC board and v1 TMC2130 installed to X axis position
#define EN_PIN    8 //enable (CFG6)
#define DIR_PIN   5 //direction
#define STEP_PIN  2 //step
#define CS_PIN    12  //chip select

bool dir = true;

#include <TMC2130Stepper.h>

TMC2130Stepper driver = TMC2130Stepper(EN_PIN, DIR_PIN, STEP_PIN, CS_PIN);

void setup() {
	Serial.begin(9600);
	while(!Serial);
	Serial.println("Start...");
  driver.begin(); 			// Initiate pins and registeries
		
  digitalWrite(EN_PIN, LOW);
  digitalWrite(DIR_PIN, LOW);
}

void loop() {
	digitalWrite(STEP_PIN, HIGH);
	delayMicroseconds(45);
	digitalWrite(STEP_PIN, LOW);
	delayMicroseconds(45);
	uint32_t ms = millis();
  static uint32_t last_time = 0;
  if ((ms - last_time) > 10000) { // using geared NEMA17 27:1 5370 steps per rev
		if (dir) {
			Serial.println("Dir -> 0");
			digitalWrite(DIR_PIN, dir);
			      
		} else {
			Serial.println("Dir -> 1");
			digitalWrite(DIR_PIN, dir);
			 
		}
		dir = !dir;
		last_time = ms;
	}
}

closed 2019-07-22 by teemuatlut