teemuatlut/TMC2130Stepper #82: stallguard2 & AccelStepper library

csurf, 2019-08-24:
I'm trying to use the TMC2130Stepper library along with the AccelStepper library in order to drive the stepper motor. The AccelStepper library requires making a function call `stepper.run()` for each step that the motor must make, which means that the motor is technically idle/stopped in between each call to the `run()` function. So, does this also mean that I won't be able to get a valid stallguard value during the off period between the calls to `stepper.run()`? Or, is it possible to get the last reading of sg_result even though the motor has just stopped? I simply need some confirmation/further explanation on how this works...

I'm thinking I'm going to need to look into using an interrupt-driven strategy to drive the motor in order to measure the load simultaneously while the motor is actually running. Any feedback/advice on getting this to work?

MilanVDB, 2020-03-29:
Working on the same thing right now. Did you find a soloution?

edwardocano, 2020-05-01:
I am working in the same thing, trying avoid using the accelstepper library. Did you sovled it @MilanVDB ?