teemuatlut/TMC2130Stepper #84: setting RMS current via SPI & via ref voltage / potentiometer, difference?

csurf, 2019-09-03:
What's the difference between setting the RMS current via SPI and setting it via the onboard potentiometer & adjusting the reference voltage?

teemuatlut, 2019-09-04:
Methodology, accuracy, ease of use, calculation, configuration, ...
You'll need to narrow down your question a bit.

csurf, 2019-09-04:
basically, I just wanted to know if there was a practical difference between the two methods, or if they accomplish the exact same goal yet via different techniques...

teemuatlut, 2019-09-04:
The main purpose would be to set the scale of the digital range.
The digital scale has 31 discrete levels and so the steps are quite coarse. However if you set the potentiometer to lets say half, and also configure the driver to use it as an input, you've then halved the full range of the digital adjustment but also have halved the size of the discrete current steps.

The way the pot is used in standalone mode is the digital current setting is set to the max, and the max is adjusted by the pot.

The way it works with pure SPI mode is the potentiometer is ignored and thus the max isn't scaled down. All the current adjustment is through the digital Current Select value.

*The TMCStepper library has no way to to know how you've set the pot so you'll end up breaking the `rms_current` methods and should set it yourself with `irun` and `ihold` methods.*
Or you can do the math behind where the current is scaled linearly according to the pot measurement.