teemuatlut/TMC2130Stepper #85: On driver board end stops

coolerooney, 2019-10-02:
Hi,

Probably I am in the wrong section- it is not an error/issue- but here goes

The Trinamic Evaluation, BOB and now also the stepsticks have pins for end stops on them (aliexpress)
How can access these end stops?
My reason for asking: have a X move of say 100 mm, but want to oscillate +/- positions on a micros scale, without interrupting the big move
Thx

Carl

teemuatlut, 2019-10-04:
On TMC2130, a diagnostic pin can be assigned to the stallGuard measurement. When the measurement is 0, the diag pin gets pulled high or low. You can then act on this signal on the MCU side.

coolerooney, 2019-10-05:
Hi Teemalut

thx for the prompt reply
I assumed that the 2130 and subsequent drivers are similar in design
on page 84 of the below 5160 document, Trinamic refers these pins as reference switches, as I understand, have no connection to stallguard- so instead of output, input

[TMC5160_Datasheet_Rev1.10.pdf](https://github.com/teemuatlut/TMC2130Stepper/files/3693390/TMC5160_Datasheet_Rev1.10.pdf)

thx

best

carel

teemuatlut, 2019-10-05:
The REFL_STEP and REFR_DIR are inputs and double as either STEP/DIR input lines or when you're using the internal ramp generator, they can act as inputs for physical reference limit switches.