teemuatlut/TMC2130Stepper #87: readme.md missing reference

i-make-robots, 2019-10-12:
"Hint: Also set DCCTRL parameters in order to operate dcStep."

no definition for DCCTRL register and api in the manual.
no definition for dcStep() ? in the manual.

teemuatlut, 2019-10-20:
To configure DCStep you can use the newer TMCStepper library which supersedes this one.

viorels, 2020-05-02:
I found in a Trinamic document that in order to use this functionality you need an extra "predriver" chip.

> The following Trinamic products (dedicated controller and dedicated driver/predriver) support dcStep in combination: TMC4361 + TMC2130

[dcStep_Basics_and_Wizard.pdf](https://www.trinamic.com/fileadmin/assets/Support/Appnotes/AN003_-_dcStep_Basics_and_Wizard.pdf) on page 4

@teemuatlut can you confirm this or did I missunderstand? Do you have any working example for this feature? Thanks