teemuatlut/TMC2130Stepper #88: Struggling with parameters

margaret-n-m, 2019-10-19:
Hi,
I'm a student working on my masters degree project, and also a newbie to programming. I've chosen TMC2130 to control a Nema 17 stepper motor for the project, as it has lots of useful functions and other advantages. I've written a simple code to control it with an Arduino M0 Pro, and when I was trying to make SPI communication work I've found Your library - it was a massive help. Now it seems that SPI is finally working, but I'm struggling with register settings. I've read the TMC2130 datasheet and tried the Initialization example parameters, they did not work. I tried quick configuration - didn't work either. I was trying my own modifications - no use. Then I tried to launch your simple.ino to check if anything happens and it worked perfectly, which means that the parameters You used would be fine for my application. I tried to check the values in crucial registers, but some of them are a write only. Could you possibly tell me what are your settings for 0x6C. 0x10, 0x11, 0x00, 0x13 and 0x70?
Margaret

teemuatlut, 2019-10-20:
The basic prerequisites for the driver to start operating on the software side are
* off time > 0
* run current > 0

The `Simple` example makes a call to the `begin` method which then pushes the default register values into the driver.
Digging through the code you can see the boot time defaults here: https://github.com/teemuatlut/TMC2130Stepper/blob/master/src/TMC2130Stepper.h#L414
```cpp
// Shadow registers
uint32_t  GCONF_sr      = 0x00000000UL,
          IHOLD_IRUN_sr = 0x00000000UL,
          TSTEP_sr      = 0x00000000UL,
          TPWMTHRS_sr   = 0x00000000UL,
          TCOOLTHRS_sr  = 0x00000000UL,
          THIGH_sr      = 0x00000000UL,
          XDIRECT_sr    = 0x00000000UL,
          VDCMIN_sr     = 0x00000000UL,
          MSLUT0_sr     = 0xAAAAB554UL,
          MSLUT1_sr     = 0x4A9554AAUL,
          MSLUT2_sr     = 0x24492929UL,
          MSLUT3_sr     = 0x10104222UL,
          MSLUT4_sr     = 0xFBFFFFFFUL,
          MSLUT5_sr     = 0xB5BB777DUL,
          MSLUT6_sr     = 0x49295556UL,
          MSLUT7_sr     = 0x00404222UL,
          MSLUTSEL_sr   = 0xFFFF8056UL,
          CHOPCONF_sr   = 0x00000000UL,
          COOLCONF_sr   = 0x00000000UL,
          DCCTRL_sr     = 0x00000000UL,
          PWMCONF_sr    = 0x00050480UL,
          tmp_sr        = 0x00000000UL,
          TPOWERDOWN_sr = 0x00000000UL,
          ENCM_CTRL_sr  = 0x00000000UL,
          GSTAT_sr      = 0x00000000UL,
          MSLUTSTART_sr = 0x00F70000UL;
```
After that an `off time` value and a typical `blank time` setting is set.
Finally the example sets the run current.

margaret-n-m, 2019-10-20:
Thank You very much! I analysed Your settings carefully and implemented them into my code - and the motor finally moved. It seems everything is working as it should be. Thank You again!
Margaret

closed 2019-10-20 by margaret-n-m