teemuatlut/TMC2130Stepper #91: Simple example only partially working

Russellfam, 2020-04-01:
Hi I am using the 'simple' example and I've tried the 'softwareSPI' example and it seems it is only partially working on the Teensy 4.0. Wiring is definitely good becuase the board I'm using I built for a dual axis telescope using OnStep and it works flawlessly there. 

functions like: rms_current, stealthChop, shaft_dir, microsteps all do nothing at all. 

For example to change direction I have to set the direction pin by digitalWrite because shaft_dir makes no change. 

I'm running this on a teensy4.0 so I would guess that might be the problem? could you suggest what I would nee to change to get this to work?

teemuatlut, 2020-04-01:
I haven't tested the Teensy 4.0 even though I have one, but my first guess is that because the software SPI is allowed to run as fast as it can, it is too fast and the clock frequency is out of spec. You'd have to artificially slow down the SW SPI implementation with delays.

Russellfam, 2020-04-01:
OK that might be the issue. I could also slow down the CPU.
 where do I put the delays?

Russellfam, 2020-04-01:
that's solved the problems it seems. dropped cpu to 150mhz from the 600 it defaults to