teemuatlut/TMC2130Stepper #95: Issue with Higher current rated motors

dev-000, 2020-08-07:
Hi,

Great job with the library!

I am playing around with TMC2130, a stepper motor and an ATSAMD21(Adafruit Feather M0 bootloader) board. 
I am able to get the motor(No load) stepping without any issue for lower currents(I<sub>RMS</sub><=872mA), the moment I go above this limit, the motor doesn't rotate.

I was able to narrow down the issue to ```rms_current()``` function in ```TMC2130Stepper.cpp```. What solved the issue for me was that I had replaced the ```else if(vsense())``` at line 365 to just ```else``` command. The ```else if``` command was not getting executed when ```CS >= 16```. But it makes no sense as to how this fixed the problem. After making the change in the library, the motor runs when I set the RMS currents more than ```872mA```.

There seems to be some issue in ```vsense()``` function? Is that case? Because when I try to read the RMS current value via rms_current() and getCurrent() I get two different values when I set the RMS current to ```872mA``` via ```driver.rms_current(1000, 0.6, 0.12);```. That can only point to the case that ```vsense()``` and ```vsense(true/false)``` is not reading/writing properly. 

I am using a ```Rsense``` resistor which is ```0.12Ω```.

Relevant initialization code to get the driver board working
```
TMC2130Stepper driver = TMC2130Stepper(nEnb, dirPin, stepPin, csPin);
driver.begin();
driver.rms_current(1000, 0.6, 0.12);  // Set stepper RMS current to 1000mA. Hold current factor: 0.6A, Rsense = 0.12Ω
driver.stealthChop(1);
```
Can you please help verify if this is an issue or if its a flaw in my logic? 

Thanking you in advance.

teemuatlut, 2020-08-07:
If CS should be below 16, then set vsense to true.
If CS should be over 16, then vsense should be false, so we test and reset it if it was set. If vsense was not set, then there's no need to write it.

Sure a mere write is faster than a read and write and that's why it was removed in TMCStepper.

My first suggestion is to move on to the newer TMCStepper library that superseded this one.
If you want to stick with this library, after setting the current, read back the chopconf and ihold_irun registers.
The motor should react to step inputs if toff > 0, irun > 0 and en pin is low.
Vsense itself doesn't prevent the motor from spinning but when reading back the value with vsense(), the read operation will overwrite the cached register value which is then modified and pushed to the register. If the read fails and reads 0, then you'll be writing toff = 0 to the driver, effectively disabling it. When you removed the vsense() call, the cached register was no longer overwritten with an invalid value and it seemed to be fixing everything.
So the likely cause for now is an issue with read operations.

dev-000, 2020-08-07:
Thanks for the fast response. 
I didn't know about the overwriting of the register on a failed read. I presume that's what is happening here. When I am reading vsense(), I was always getting a zero. 

I had tried the newer TMCStepper library but couldn't get it working with my setup. But I didn't spend any time trying to figure out why it wasn't working. Let me try the new library once again and report back. If it works, I can get going with that. I presume the same initialization commands and functions should work for the new library as well?

dev-000, 2020-08-07:
Migrated to the [new TMCStepper library](https://github.com/teemuatlut/TMCStepper). Got it working fine with my setup. 
Good that you added the note to ReadMe to help anyone use the newer library from now. 

Thanks again for all the help. 😃

dev-000, 2020-08-07:
Seems the same issue of ```vsense``` persists in the newer library too.  

For I<sub>RMS</sub><=872mA(CS<16 limit for ```Rsense = 0.12Ω```), ```driver.rms_current()``` still returns a wrong value. ```vsense``` seems to be stuck at zero. 
But the motor runs in both cases though. 

I just saw that someone else has opened an [issue](https://github.com/teemuatlut/TMCStepper/issues/126) on this same problem in the new library too. I will post there from now on if there is a fix.

teemuatlut, 2020-08-07:
`Arduino Nano 33 IOT` with TMCStepper@master
```cpp
#include <TMCStepper.h>

#define EN_PIN            3 // Enable
#define STEP_PIN          2 // Step
#define CS_PIN           10 // Chip select
#define R_SENSE 0.11f // Match to your driver

// Select your stepper driver type
TMC2130Stepper driver(CS_PIN, R_SENSE);                           // Hardware SPI

void setup() {
  delay(2000);
  pinMode(EN_PIN, OUTPUT);
  pinMode(STEP_PIN, OUTPUT);
  digitalWrite(EN_PIN, HIGH);
  Serial.begin(9600);
  while(!Serial);
  Serial.println("########## " __TIME__ " ########");

  SPI.begin();
  driver.begin();

  driver.rms_current(600);        // Set motor RMS current
  Serial.print("vsense = ");
  Serial.print(driver.vsense());
  Serial.print("\trms_current = ");
  Serial.print(driver.rms_current());
  Serial.print("\tirun = ");
  Serial.println(driver.irun());

  driver.rms_current(1000);        // Set motor RMS current
  Serial.print("vsense = ");
  Serial.print(driver.vsense());
  Serial.print("\trms_current = ");
  Serial.print(driver.rms_current());
  Serial.print("\tirun = ");
  Serial.println(driver.irun());
}

void loop() {}

```
For me returns:
```code
########## 23:44:04 ########
vsense = 1	rms_current = 581	irun = 18
vsense = 0	rms_current = 994	irun = 17
```

dev-000, 2020-08-08:
Thanks for taking the time to check this.
I tried your same code on my first setup(Custom TMC2130 + Atsamd21 board) changing only the pin numbers. I got the following

```vsense = 0  rms_current = 1077  irun = 20```
```vsense = 0	rms_current = 974	irun = 18```

which corresponds to the fact that vsense is not getting set(or not getting read properly) 

To explore why this was happening, I went back to a normal Atsamd21 M0 Adafruit feather board and original TMC SilentStepStick SPI board setup. Made wire connections between them. For the same connections and the same code, I get the output the same as you reported.

```vsense = 1	rms_current = 581	irun = 18```
```vsense = 0	rms_current = 994	irun = 17```

It seems then that the TMCStepper library is working and I am starting to suspect that it might be a hardware issue in my custom board. I don't think its an SPI connection issue as its soldered on. Now when I am rechecking the TMC SilentStepStick schematic and mine(Attached below), seems that I left ``` AIN_IREF ``` open. But I am unsure if that will cause this issue which we are discussing. 
[image]
Any tips on debugging this issue on my custom board? It's a TQFP48 package. I am unclear why there might be a read/write issue (as the motor is configured and running).

teemuatlut, 2020-08-08:
Hardware design is not really my strong suite. My first thought was the unconnected SPI_MODE pin but according to the datasheet that should have a pullup resistor and so should be fine. Maybe you can try pulling it to 3.3 none-the-less.

AIN_VREF is only used if I_scale_analog is enabled in gconf register.

Can you successfully read the IOIN register?

dev-000, 2020-08-08:
I tried reading IOIN register with ``` driver.IOIN() ``` but it keeps returning zero. So I am guessing it's not reading(It's reading fine in Feather M0 + TMCsilentstep setup). 
I am unclear as to how a write( for configurations) are possible and yet read keeps failing. 
Regarding the 3V3 pullups, I didn't give that for the same reasons you mentioned. I am unsure about AIN_VREF though as there is no pullup there. But I am using that in my code also. 
I have to really dig deep into checking what's wrong.

closed 2023-07-05 by dev-000