teemuatlut/TMC2130Stepper #96: Possible pull request... Use MCP23017 to multiple SPI TMC2130

julioj4pa, 2020-10-20:
Hi nice to e-meet you.
I'm newbie at github... By the way thank you for every developer of this library it helps me a lot. I'm working on project to against hunger.
So I changed library to use MCP23017 IO expander to use less DIO when someone need to use multiple SPI TMC2130 with AccelStepper.
If someone needs it. I can zip it and send with example...

TMC2130Stepper driver = TMC2130Stepper(EN_PIN, DIR_PIN, STEP_PIN, CS_PIN, 51, 50, 52);
driver.addMcp(&mcp);
 driver.begin(); 

void (TMC2130Stepper or AccelStepper)::addMcp(MCP23017 *nMcp){
	mcp=nMcp;
	useMcp=true;
}

Inside Library function... Example
if(useMcp){
	mcp->digitalWrite(_pinCS, HIGH);
}
else{
	digitalWrite(_pinCS, HIGH);
}

teemuatlut, 2020-10-23:
You can fork it and that way have it available for others.
I won't be developing this library much further as it has been superseded by the TMCStepper lib.

julioj4pa, 2020-10-23:
Ok. Thank you

closed 2020-10-23 by julioj4pa

Er9y714, 2020-12-17:
@julioj4pa can I get a copy of library please? 
I needed this[ long ago.](https://forum.arduino.cc/index.php?topic=667908.msg4518856#msg4518856) Then I figured to connect 2 motors on the same ESP8266 by using SPI. Nevertheless, this library might be useful someday.

teemuatlut, 2020-12-17:
With TMCStepper you can daisy-chain SPI based drivers and control 127 of them with just the four pins.

Er9y714, 2020-12-17:
@teemuatlut , but do not we need separate pins for each driver on the STEP and CS pins? Then for each driver, we are using up more pins.

teemuatlut, 2020-12-18:
You share a single CS pin between all the drivers and shift data through driver to address one further down the chain.
You'll need to use something like the 5130 or 5160 and their internal step generator if you want to remove the step pin.

Er9y714, 2020-12-18:
I am not familiar of these methods. I will need to do some research. Thanks for the tip!