teemuatlut/TMC2130Stepper #97: Is DIR pin obsolete? / verify current fraction of steps / 1 step function in SPI

Er9y714, 2020-12-17:
When using driver.begin();  function, I suppose the driver is connected via SPI. Then we do not need DIR pin. I verified this by not connecting the pin to anywhere. driver.shaft_dir(0);  still changes the direction. 

In this case what is the point of using DIR_PIN in this line?
TMC2130Stepper driver = TMC2130Stepper(EN_PIN, **DIR_PIN,** STEP_PIN, CS_PIN, MOSI_PIN, MISO_PIN, SCK_PIN);     

When I remove the DIR_PIN (to save a pin on the board), the function does not work of course. 

My first question is, if I will use SPI connection, how can I not use DIR_PIN for this function? I do not want to waste a pin that is not being used. 

------

Second question is, is there a way to know where the step is in terms of fractions (instead of counting from the start)? Imagine I advance step by step (each step is 1/8 step) and after 3 steps I would like to check what the fraction it is standing. 

-----

Is there an SPI function for 1 step advancing instead of setting STEP_PIN as HIGH and then LOW?

Thanks!

teemuatlut, 2020-12-17:
Basically yes, it's obsolete and the library doesn't do anything else with it other than setting it as an output. I had some plans to use the step, dir and en pins but there are better libraries to handle the step generation. I also moved to developing [a new library](https://github.com/teemuatlut/TMCStepper) with new architecture and the unnecessary pins were dropped from the API.

You can read current microstep counter with `.MSCNT()`.

To take a single microstep you'll have to use the max 256 microstep setting.

Er9y714, 2020-12-17:
Wow thank you for prompt response!

I have come across with that library but I did not know they were different. Because I have spent so much time to figure this library out, I do not think I will try other libraries. It is silent while holding the motor, I can adjust the microstep setting by software and it works :).

It seems I will not be able to use the step function because of 256 limitation. 

Have a nice evening!

closed 2020-12-17 by Er9y714