teemuatlut/TMC2130Stepper #100: Motor won't rotate if stealthChop isn't enabled

Robin-12, 2021-04-20:
Hi,

First of all, great work on the library! It has been very helpful!
I am trying to operate a NEMA17 motor and need to know when it is stalled, hence i chose the SilentStepStick TMC2130. I modified the pins and got the simple example to work. However, when uploading the stallGuard.ino example to my Arduino Uno (clone) the motor won't rotate. 

While trying to figure out why simple.ino works and stallGuard.ino doesn't, I noticed that if i put `driver.stealthChop(1);` in the stallGuard example it does rotate. However, when doing this, `driver.sg_result();` only returns zero. By looking through the issues of this library, i found that this is because you cannot read sg_result when stealthChop is enabled. 

So here is **my question**: Does anybody have any idea why I cannot get the motor to rotate without stealthChop?

Is it because my TMC2130 is configured in SPI mode? (open pads, not soldered)

Any help is much appreciated!
Robin

Here is the rest of the code where i changed 3 things form the original example:

1. Modified the pins
2. Modified the PORT
3. Added stealthChop(1) at the end of TMC2130 config

```cpp
#define MAX_SPEED  40 // In timer value
#define MIN_SPEED  1000

#define STALL_VALUE 0 // [-64..63]

// Note: You also have to connect GND, 5V and VM.
//       A connection diagram can be found in the schematics.
#define EN_PIN    5  // Nano v3:  16 Mega:  38  //enable (CFG6)
#define DIR_PIN   8  //           19        55  //direction
#define STEP_PIN  6  //           18        54  //step
#define CS_PIN    9  //           17        64  //chip select

#define BAUDRATE 9600
#include <TMC2130Stepper.h>
#include <TMC2130Stepper_REGDEFS.h>

TMC2130Stepper driver = TMC2130Stepper(CS_PIN);

bool vsense;

uint16_t rms_current(uint8_t CS, float Rsense = 0.11) {
  return (float)(CS+1)/32.0 * (vsense?0.180:0.325)/(Rsense+0.02) / 1.41421 * 1000;
}

void setup() {
  //init serial port
  {
    Serial.begin(BAUDRATE); //init serial port and set baudrate
    while(!Serial); //wait for serial port to connect (needed for Leonardo only)
    Serial.println("\nStart...");
    pinMode(EN_PIN, OUTPUT);
    pinMode(DIR_PIN, OUTPUT);
    pinMode(STEP_PIN, OUTPUT);
    pinMode(CS_PIN, OUTPUT);
    digitalWrite(EN_PIN, HIGH); //deactivate driver (LOW active)
    digitalWrite(DIR_PIN, LOW); //LOW or HIGH
    digitalWrite(STEP_PIN, LOW);
    digitalWrite(CS_PIN, HIGH);
    SPI.begin();
    pinMode(MISO, INPUT_PULLUP);
  }

  //set TMC2130 config
  {
    driver.push();
    driver.toff(3);
    driver.tbl(1);
    driver.hysteresis_start(4);
    driver.hysteresis_end(-2);
    driver.rms_current(600); // mA
    driver.microsteps(16);
    driver.diag1_stall(1);
    driver.diag1_active_high(1);
    driver.coolstep_min_speed(0x00); // 20bit max
    driver.THIGH(0);
    driver.semin(5);
    driver.semax(2);
    driver.sedn(0b01);
    driver.sg_stall_value(STALL_VALUE);
    driver.stealthChop(1);
  }

  // Set stepper interrupt
  {
    cli();//stop interrupts
    TCCR1A = 0;// set entire TCCR1A register to 0
    TCCR1B = 0;// same for TCCR1B
    TCNT1  = 0;//initialize counter value to 0
    OCR1A = 256;// = (16*10^6) / (1*1024) - 1 (must be <65536)
    // turn on CTC mode
    TCCR1B |= (1 << WGM12);
    // Set CS11 bits for 8 prescaler
    TCCR1B |= (1 << CS11);// | (1 << CS10);  
    // enable timer compare interrupt
    TIMSK1 |= (1 << OCIE1A);
    sei();//allow interrupts
  }

  //TMC2130 outputs on (LOW active)
  digitalWrite(EN_PIN, LOW);

  Serial.println(driver.version(),BIN);
  vsense = driver.vsense();
}

ISR(TIMER1_COMPA_vect){
  PORTD |= 1 << 6;
  PORTD &= ~(1 << 6);
}

void loop()
{
  static uint32_t last_time=0;
  uint32_t ms = millis();

  while(Serial.available() > 0) {
    int8_t read_byte = Serial.read();
    if (read_byte == '0')      { TIMSK1 &= ~(1 << OCIE1A); digitalWrite( EN_PIN, HIGH ); }
    else if (read_byte == '1') { TIMSK1 |=  (1 << OCIE1A); digitalWrite( EN_PIN,  LOW ); }
    else if (read_byte == '+') if (OCR1A > MAX_SPEED) OCR1A -= 20;
    else if (read_byte == '-') if (OCR1A < MIN_SPEED) OCR1A += 20;
  }
   
  if((ms-last_time) > 100) //run every 0.1s
  {
    last_time = ms;
    uint32_t drv_status = driver.DRV_STATUS();
    Serial.print("0 ");
    Serial.print((drv_status & SG_RESULT_bm)>>SG_RESULT_bp , DEC);
    Serial.print(" ");
    Serial.println(rms_current((drv_status & CS_ACTUAL_bm)>>CS_ACTUAL_bp), DEC);
  }
}
```

teemuatlut, 2021-04-20:
Switching between stealthChop and spreadCycle shouldn't cause your motor to stop turning. The TMC2130 needs a couple of things to be enabled; EN line needs to be pulled low, off time needs to be not zero and there should be no fault conditions.
All these conditions should be fulfilled by your code.
I uploaded your code with the required changes to my pins, the step pin port and the pin in port. I was using a RAMPS board so my step pin was on PORTL3.

---
[image]
```cpp
#define MAX_SPEED  40 // In timer value
#define MIN_SPEED  1000

#define STALL_VALUE 0 // [-64..63]

// Note: You also have to connect GND, 5V and VM.
//       A connection diagram can be found in the schematics.
#define EN_PIN    62 // Nano v3:  16 Mega:  38  //enable (CFG6)
#define DIR_PIN   48 //           19        55  //direction
#define STEP_PIN  46 //           18        54  //step
#define CS_PIN    53  //           17        64  //chip select

#define BAUDRATE 9600
#include <TMC2130Stepper.h>
#include <TMC2130Stepper_REGDEFS.h>

TMC2130Stepper driver = TMC2130Stepper(CS_PIN);

bool vsense;

uint16_t rms_current(uint8_t CS, float Rsense = 0.11) {
  return (float)(CS+1)/32.0 * (vsense?0.180:0.325)/(Rsense+0.02) / 1.41421 * 1000;
}

void setup() {
  //init serial port
  {
    Serial.begin(BAUDRATE); //init serial port and set baudrate
    while(!Serial); //wait for serial port to connect (needed for Leonardo only)
    Serial.println("\nStart...");
    pinMode(EN_PIN, OUTPUT);
    pinMode(DIR_PIN, OUTPUT);
    pinMode(STEP_PIN, OUTPUT);
    pinMode(CS_PIN, OUTPUT);
    digitalWrite(EN_PIN, HIGH); //deactivate driver (LOW active)
    digitalWrite(DIR_PIN, LOW); //LOW or HIGH
    digitalWrite(STEP_PIN, LOW);
    digitalWrite(CS_PIN, HIGH);
    SPI.begin();
    pinMode(MISO, INPUT_PULLUP);
  }

  //set TMC2130 config
  {
    driver.push();
    driver.toff(3);
    driver.tbl(1);
    driver.hysteresis_start(4);
    driver.hysteresis_end(-2);
    driver.rms_current(600); // mA
    driver.microsteps(16);
    driver.diag1_stall(1);
    driver.diag1_active_high(1);
    driver.coolstep_min_speed(0x00); // 20bit max
    driver.THIGH(0);
    driver.semin(5);
    driver.semax(2);
    driver.sedn(0b01);
    driver.sg_stall_value(STALL_VALUE);
    driver.stealthChop(0);
  }

  // Set stepper interrupt
  {
    cli();//stop interrupts
    TCCR1A = 0;// set entire TCCR1A register to 0
    TCCR1B = 0;// same for TCCR1B
    TCNT1  = 0;//initialize counter value to 0
    OCR1A = 256;// = (16*10^6) / (1*1024) - 1 (must be <65536)
    // turn on CTC mode
    TCCR1B |= (1 << WGM12);
    // Set CS11 bits for 8 prescaler
    TCCR1B |= (1 << CS11);// | (1 << CS10);  
    // enable timer compare interrupt
    TIMSK1 |= (1 << OCIE1A);
    sei();//allow interrupts
  }

  //TMC2130 outputs on (LOW active)
  digitalWrite(EN_PIN, LOW);

  Serial.println(driver.version(),BIN);
  vsense = driver.vsense();
}

ISR(TIMER1_COMPA_vect){
  PORTL |= 1 << 3;
  PORTL &= ~(1 << 3);
}

void loop()
{
  static uint32_t last_time=0;
  uint32_t ms = millis();

  while(Serial.available() > 0) {
    int8_t read_byte = Serial.read();
    if (read_byte == '0')      { TIMSK1 &= ~(1 << OCIE1A); digitalWrite( EN_PIN, HIGH ); }
    else if (read_byte == '1') { TIMSK1 |=  (1 << OCIE1A); digitalWrite( EN_PIN,  LOW ); }
    else if (read_byte == '+') if (OCR1A > MAX_SPEED) OCR1A -= 20;
    else if (read_byte == '-') if (OCR1A < MIN_SPEED) OCR1A += 20;
  }
   
  if((ms-last_time) > 100) //run every 0.1s
  {
    last_time = ms;
    uint32_t drv_status = driver.DRV_STATUS();
    Serial.print("0 ");
    Serial.print((drv_status & SG_RESULT_bm)>>SG_RESULT_bp , DEC);
    Serial.print(" ");
    Serial.println(rms_current((drv_status & CS_ACTUAL_bm)>>CS_ACTUAL_bp), DEC);
  }
}
```

Robin-12, 2021-04-21:
Thanks for your quick reply! Hmm, I cannot find what's wrong with my setup, i feel like i have checked everything 100 times. What brand of driver board are you using? 
Or could the fact that i'm not using RAMPS have anything to do with it?
When I disable stealthChop, the motor just stops turning and I can hear some crackling and high pitched sounds. I have tried numerous configurations of toff, tbl, hstrt and hend but none of them could get it to turn. 

If anybody has any suggestions, I would love to hear :)

Robin-12, 2021-04-21:
Read out the DRV_STATUS and the overtemperature pre-warning threshold (opwt) and overtemperature limit (ot) flags are high which disable the driver. However, this happens upon start while I actively cool the driver with a heatsink and fan. I also set the current to 600mA using `rms_current(600)`. I am clueless as to why this flag goes up.